waveform80/picamera #348: H264 encoder limits?

g40, 2016-11-24:
The documentation hints that it might be possible to encode video at greater than 1920x1080 with a PI3/V2 combination. Can anyone share numbers if they have managed to do this? Many thanks.

6by9, 2016-11-24:
Which documentation? Link please.
2,048×1,024 @ 30fps should work, but not sure if you count that as greater when it is only in one direction.

(Please refer to https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels for details of H264 levels).

The codec block is specified to H264 level 4.0 (245,760 macroblocks/second), although it will do bitrates up to about 30Mbit/s which pushes it into level 4.1 territory (level 4.1 ONLY increases the max bitrate).

Level 4.0 specifies a maximum of 8,192 macroblocks/frame, which is where the resolution limit comes from. Macroblocks are 16x16. (2048/16) * (1024/16) = 8192 or (1920/16) * (1088/16) = 8160.
My understanding is that there is a hardware image buffer within the codec block which is wide enough for level 4.0 (ie 2048 pixels) and no more. You have to go up to level 5.0 to get increased resolutions, and that is definitely not supported.

So no you can't increase resolution.
People have overclocked the H264 block to achieve higher frame rates to go alongside the 720P high frame rate mode of the V2 camera. You have to select level 4.2 first.

g40, 2016-11-24:
Excellent answer, thank you.