waveform80/picamera #350: Camera hangs up unexpectedly

BruXy, 2016-11-29:
Hi all, I am using PiCamera 1.12 for creating day timelapse, here is the code:

https://github.com/BruXy/time-lapse-utils/blob/master/snapshot.py

However, after almost a month of continuous usage, I have realized that camera hangs up (yesterday twice) and I need to kill the process with SIGTERM and restart RaspberryPi to be able to use the camera again.

When I tested with pdb, the process hang at  the initialization `with picamera.PiCamera() as camera:`. The CPU load remained normal (strace raspivid stopped on a futex syscall, but I did not save the output).

I did some brief system check like dmesg, /var/log/messages, but I did not find anything relevant.

I am worried, that it will happen again. Is there any option to check camera state or enable some debug mode to see what it going on?

Regards,
BruXy

BruXy, 2017-03-10:
Since the report of issue the situation did not happen again. Actually, I am not sure if this was happening for version 1.12, because during problem debugging I probably did pip upgrade.

closed 2017-03-10 by BruXy

simbalsyd, 2021-01-18:
I have exactly the same problem with version 1.13. Occasionally the camera hangs and when I try to debug it hang at the initialization.