waveform80/picamera #352: PiMotionAnalysis sensitivity

andrhahn, 2016-12-17:
Hi, I am writing a motion detector using PiMotionAnalysis following this example:  https://picamera.readthedocs.io/en/release-1.12/recipes2.html#custom-outputs

Motion is detected at intermittent intervals, approximately every 10-20 seconds even though nothing has appeared to change.

Is this likely due to reflection [(347)](https://github.com/waveform80/picamera/issues/347) or is there something else I should look at?  This seems to happen in several conditions (morning, afternoon, evening, low light, lots of light)

If reflection, would you have a generic code example in mind?

Thanks-

ras-marques, 2016-12-17:
Hello,  it is entirely possible that automatic adjustments of the camera parameters (white balance and exposure times IIRC) change the values of many pixels at once, therefore rising the "motion amount" to a level at which it is understood by your algorithm as effective motion.

I am writing this after having some problems inside a room with poor illumination, in which picamera had difficulties choosing adequate levels (of what, I can't remember), and that screwed the motion capture algorithm I was developing big time. Fixing everything by hand solved the problem and I could work on that.

Does this help?

andrhahn, 2016-12-18:
Hi thanks I will try your suggestions.  I am relatively new to photography and video so still have a lot to learn.  After watching back a video triggered by one of these "false positives" it looks like an exposure issue, if that makes sense.  The frames in the video change noticeably.  I might try a new v2 camera module to rule out defective hardware as well.  I'll post back if that fixes things.  It's been a fun project so far.  Thanks again

FutureSharks, 2017-09-03:
@ras-marques can you show exactly what you set to disable automatic adjustments?

I set some settings here:
https://github.com/FutureSharks/rpi-security/blob/master/rpisec/rpis_camera.py#L139-L147
But I still get false positive motion detected regularly.

ras-marques, 2017-09-03:
Are you using the latest picamera version? Up until 1.9 you had to add a 2 seconds delay or something to let the camera adjust first and then disable the auto mode, like so:

`camera.start_preview()
print "camera adjusting"
time.sleep(2) #
print "camera adjusted"
camera.awb_mode = 'off'
camera.awb_gains = (41/32,221/128)
camera.shutter_speed = 30000
camera.exposure_mode = 'off'`

Check [#171](https://github.com/waveform80/picamera/issues/171)

Even if you don't have an ancient version from the begining of 2015, those settings may help. What I did IIRC, was let the camera adjust automatically, read the result of the camera.awb_gains and then fix it in the code. Also tried doing that on runtime but somehow I didn't succeed, don't know why...

Post back your results, I don't play with this for a long time.

FutureSharks, 2017-09-03:
I'm using version 1.13.

In [my application](https://github.com/FutureSharks/rpi-security), the camera is used for taking photos too so motion detection would rarely ever run from a cold start. Before motion detection starts (`camera.start_recording`), I just [freeze all the settings](https://github.com/FutureSharks/rpi-security/blob/master/rpisec/rpis_camera.py#L139-L147v).

I don't really know what else to try. I get false positives once every 1-3 days. Perhaps the method of detecting by using numpy arrays is just too crude.