waveform80/picamera #353: Recording and Streaming using picamera and pistreaming

survinderpal, 2016-12-23:
I am trying to record hd video using picamera and simultaneously stream it to the other end using pistreaming in low resolution. In pistreaming, the data is captured as 'yuv'. If I record in the same format, the memory fills up quite quickly. Is there any way to make both recording and streaming simultaneously on Pi? Is it possible to encode in h264 for recording and send in yuv simultaneously?

waveform80, 2017-01-16:
Certainly: have a look at the [recording multiple resolutions](http://picamera.readthedocs.io/en/release-1.12/recipes2.html#recording-at-multiple-resolutions) recipe which demonstrates recording two different H264 streams at different resolutions. That same technique can be used with different formats as well.

houssammoncef, 2017-04-12:
hello i want to change the part of vlc with cv2 or other stuff to display it  in gui
and thanks for your advices

AndreAhmed, 2019-01-28:
@waveform80  I have the same issure. I'm using picstreaming and picamera. But I want BOTH to be at h264
which stuff needs to be changed in picstreaming to make it work with h264