waveform80/picamera #355: TypeError: __class__ set to <class 'picamera.mmal.MMAL_FOURCC_T'> defining 'MMAL_FOURCC_T' as <class 'picamera.mmal.MMAL_FOURCC_T'>

robmarkcole, 2016-12-28:
I have a script: 

import picamera
camera = picamera.PiCamera()
camera.vflip = True
camera.capture('new_image3.jpg')

when run I get the following error:

Traceback (most recent call last):
  File "camera_test.py", line 1, in <module>
    import picamera
  File "/usr/local/lib/python3.6/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/local/lib/python3.6/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/local/lib/python3.6/site-packages/picamera/mmal.py", line 153, in <module>
    class MMAL_FOURCC_T(ct.c_uint32):
TypeError: __class__ set to <class 'picamera.mmal.MMAL_FOURCC_T'> defining 'MMAL_FOURCC_T' as <class 'picamera.mmal.MMAL_FOURCC_T'>

After some googling I am none the wiser, any advice appreciated.

myidisnotthis, 2016-12-31:
I ran into the same problem running the picamera quick-start code. Also using Python3.6

But OK when I ran in Pyhton3.4

waveform80, 2017-01-12:
Hmm, interesting. Can't say I've touched 3.6 yet. I'll have a look into this, but no promises for 1.13 (3.6 is *very* new and isn't likely to appear in regular Raspbian for a long time yet so this isn't a huge priority).

waveform80, 2017-01-12:
I haven't tested it yet (I'd need to build a py3.6 installation first) but just skimming the ["What's New in 3.6" list](https://docs.python.org/3/whatsnew/3.6.html) the following item stands out: [PEP487: Simpler customization of class creation](https://docs.python.org/3/whatsnew/3.6.html#pep-487-simpler-customization-of-class-creation). I wouldn't be overly surprised if `ctypes` was doing something funky in class creation that causes some issue when `ctypes` stuff is sub-classed in 3.6.

waveform80, 2017-01-13:
Finally narrowed this down after managing to get py3.6 built on Raspbian: the issue is the `super()` call in `__eq__`. Seems like a `super()` call in any `ctypes` sub-class fails in 3.6. The following minimal test case demonstrates this:

```python
import ctypes as ct

class TestSubclass(ct.c_uint32):
    def __repr__(self):
        return super().__repr__()
```

This fails on 3.6 but works on all prior versions (testing 2.7, 3.2, 3.4, and 3.5). Still not sure precisely what change in 3.6 is responsible for this, although my money's on the PEP487 implementation. Time to read that, but at the moment I'm thinking this is an upstream Python issue (unless my investigation of PEP487 shows I'm doing something terribly heinous by calling `super` in a `ctypes` subclass).

waveform80, 2017-01-13:
Having read through PEP487 I'm baffled; can't see anything in there (or the associated patches) that deals with `super` or setting `__class__` so I've opened [issue 29270](http://bugs.python.org/issue29270) upstream on CPython.

waveform80, 2017-01-15:
This is an issue in Python 3.6 itself. I'll keep an eye on Python issue 29270 and if a fix for that isn't released prior to the next picamera release, I'll stick a workaround in picamera (which isn't terribly difficult as that's the only place super() is used in mmal.py).

crossi36, 2017-01-15:
Have you actually implemented the workaround yet? If so, could you post it here so we can manually change the required files until there is a solution that doesn't involve tinkering with the files yourself?

waveform80, 2017-01-15:
Okay, here's the simplest patch that'll get things working for 1.12 on Python 3.6:

```patch
diff --git a/picamera/mmal.py b/picamera/mmal.py
index cd77362..2bd92c7 100644
--- a/picamera/mmal.py
+++ b/picamera/mmal.py
@@ -152,10 +152,7 @@ MMAL_TIME_UNKNOWN = ct.c_int64(1<<63)
 
 class MMAL_FOURCC_T(ct.c_uint32):
     def __eq__(self, other):
-        if isinstance(other, int):
-            return self.value == other
-        else:
-            return super(MMAL_FOURCC_T, self).__eq__(other)
+        return self.value == other
 
     def __str__(self):
         return ''.join(chr(self.value >> i & 0xFF) for i in range(0, 32, 8))
```

It's not great and I'll merge something a bit better for 1.13 (assuming upstream's not fixed it by then; I'm monitoring the ticket, but right now it's frankly above my head as it's dealing with Python internals beyond the level I've dealt with at this point: I'll try and find some time to catch up!). Anyway, if you're on Arch and using 3.6, apply this patch to your installation and you should be good to go (I've run the test suite against this with a Python 3.6 install on Raspbian).

waveform80, 2017-01-28:
569463f87782b240fbc6b99de647eb3f0de2655c and the few commits after it should fix this for the next release.

I figure sub-classing `ctypes` is evidently risky so this does the "simple" thing and reverts MMAL encodings to straight ctypes integer representations. Unfortunately it does mean for anyone using the `mmalobj` layer, that encodings no longer have a "nice" `repr` output - they're just big numbers. Call the new `FOURCC_str` function to convert encoding integers to a string (e.g. for command line interrogation or debugging purposes). The various repr's of things like ports, and pipeline dumps will implicitly do this (so most people will probably never need `FOURCC_str`).