waveform80/picamera #359: Fail to preview at full resolution

pirquessa, 2017-01-21:
I try to display a preview with my RPI3 & Camera V2 in (3280, 2464) resolution. It works for a few images, then screen become black.

Doc mention:
> The maximum resolution of the V2 camera can cause issues with previews. Currently, picamera runs previews at the same resolution as captures (equivalent to -fp in raspistill). You may need to increase gpu_mem in /boot/config.txt to achieve full resolution operation with the V2 camera module.

I would like to know if someone make it works. I try to set 512Mo memory for the GPU, it doesn't help.

If it is not possible, what is the faster way to change resolution ? Currently I do something like that:

`		self.camera.stop_preview()

		self.camera.resolution = (3280, 2464)
		self.camera.framerate = 2

		# Take picture & display it
		logger.debug("Take picture")
		self.camera.capture(self.stream, format=self.config.photoFormat)
		self.stream.seek(0)
		img = Image.open(self.stream)

		# Switch back to video mode, then restart preview
		self.camera.resolution = (1640, 922)
		self.camera.framerate = 32

		self.camera.start_preview()`

But it took 5 seconds to take the picture and display again the preview...

waveform80, 2017-01-21:
See [FAQ 15](http://picamera.readthedocs.io/en/release-1.12/faq.html#preview-flickers-at-full-resolution-on-a-v2-module). Simply put: the V2 camera module's native resolution is too high to run the preview at, but then there's little point actually running the preview at the native resolution (under the assumption your monitor isn't capable of displaying 3280x2464 pixels). Just pick a resolution that your monitor can do, which has the same aspect ratio as the capture resolution and set the preview resolution accordingly, e.g.

```python
camera = PiCamera(resolution=(3280, 2464))
camera.start_preview(resolution=(1440, 1080))
```

It'll still capture at the full resolution, but the preview will be scaled down to something sensible. Oh, and you can set your GPU mem back to 128Mb too :)

pirquessa, 2017-01-22:
WWWWWHHHHHHAAAAAAAAATTTTTTTTTTTTTTTTTTTTTTTTT !!!!

That's soooooooo cooooollllllllll :) Thx

Maybe you should add the options to the doc of [start_preview](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.start_preview), I did'nt see this option in the doc. Is there any other options ? I didn't saw the FAQ 15 :-(

pirquessa, 2017-01-22:
So cool, I just test your proposal. It only take a second to take the picture and display again the preview...

If you have any tip to improve again this timing, your welcome ! :)

waveform80, 2017-01-23:
> Maybe you should add the options to the doc of start_preview, I did'nt see this option in the doc. Is there any other options ? I didn't saw the FAQ 15 :-(

No prob :)

As to other options this is indicated (admittedly rather obscurely) in [the documentation](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.start_preview): "A PiRenderer instance (more specifically, a PiPreviewRenderer) is constructed *with the keyword arguments captured in **options***"; in other words whatever options you specify as keyword arguments to `start_preview` are passed onto the [`PiPreviewRenderer` constructor](http://picamera.readthedocs.io/en/release-1.12/api_renderers.html#picamera.PiPreviewRenderer). If you look at that bit of the document you'll see it takes numerous parameters including `resolution`, `layer`, `alpha`, `fullscreen` and so on. The meaning of those settings can be found in the `PiPreviewRenderer` documentation, and its parent class [`PiRenderer`](http://picamera.readthedocs.io/en/release-1.12/api_renderers.html#picamera.PiRenderer) (the reason for the parent class is that renderers are used to implement both the camera preview and preview overlays).

As for speeding up captures, if you want full quality then no, there's no way to speed up the capture. It's relatively slow because the camera switches modes when capturing stills. If you specify `use_video_port=True` you'll get a much faster capture because you avoid the mode switch, but the quality will be considerably lower. There's a lot of detail on what goes on behind the scenes in the [camera hardware chapter](http://picamera.readthedocs.io/en/latest/fov.html) of the docs (that link goes to the latest, still unfinished version - it's probably quite rough in some places, but there's a lot of useful stuff in there).

pirquessa, 2017-01-23:
Thx for the doc. That's **perfect** :)

Your right, I try to get the best pictures I can. Taking the picture take ~650ms, creating a preview from this high quality image cost ~650ms. My question was just: "Is there any tip I should know to gain some ms ?" ;-)

waveform80, 2017-01-23:
Ah, for thumbnail previews when you're capturing a JPEG the camera automatically produces a thumbnail preview JPEG and embeds it in the EXIF data of the captured JPEG. The parameters for the thumbnail (size and quality) can be controlled with *thumbnail* argument of the `capture` method (there are defaults but I don't recall them off the top of my head).

To extract the thumbnail you'll need something that can read EXIF data - I think there's a part of the [PIL](https://pillow.readthedocs.io/) library (`sudo apt-get install python-imaging python3-imaging`) that'll do that but again my memory is fuzzy!

Unfortunately thumbnails are not capable of being produced by the other image codecs (PNG, etc.) and neither can the MJPEG codec which is used for JPEG captures from the video port but I don't think those apply here. Of course, the other option is to capture the image without a thumbnail then post-process it yourself (scaling it down) to produce the thumbnail manually. That's also pretty easy with PIL.

pirquessa, 2017-01-23:
I'm currently using the second solution you propose with PIL:

- Take the big picture
- Post process (resize & crop)
- Display

It take ~650ms

I'll try the thumbnail solution :)

And improve the current one with maybe disabling the embedded thumbnail that I currently don't use to save some ms during capture...

waveform80, 2017-01-23:
Disabling the embedded thumbnail will save very little time; it's generated by the GPU (which has hardware acceleration for encoding JPEGs). I'd be quite surprised if disabling it shaved more than a few ms off any capture time.

pirquessa, 2017-01-23:
Ok, so your other solution (exporting the main image with a medium thumbnail (1000px wide) then extract it) must be better than mine. Can't wait to test it !

pirquessa, 2017-01-23:
:(

That's not that easy to get the thumbnail from the Exif data. I give up for today. If someone have an exemple, I would be happy to get it.

waveform80, 2017-01-24:
If you're not used to parsing binary files, this can be quite tricky. I've dug out and polished up some code I wrote ages ago for picroscopy and never got around to merging (I think I wanted to remove the reliance on the external exiftool calls, but PIL's Exif parsing wasn't good enough to extract the thumbnails - or something like that, it's been a few years!).

Anyway, this should do the job (tested on a V1 module, but should work with either I think):

https://gist.github.com/waveform80/a699c495126559d5a0f494fd53065763

pirquessa, 2017-01-24:
Did you just write/test this for me ? Are you trying to get a date with me ? :-p

[image]

That's so kind of you. I'll try this tonight !

Can you explain l130/131 what are the indexes 513 & 514 ? You use what you print to find the index of the thumbnail ?

pirquessa, 2017-01-24:
It works very well !!!! It's very very fast !!!!

BUT... The quality is very poor :(

What are the allowed values for the resolution and the quality of the thumbnail ? It looks like I cannot use a width > 1024. I don't find the limit for quality but 1000 is not really better than 100.

waveform80, 2017-01-24:
No, not just for you; picroscopy's an old project I've not got around to finishing (yet) that (ideally) needed to pull out the thumbnail data for display in the web interface. I think the committed version on github still uses exiftool (an external binary) for pulling out the thumbnails but that's *horribly* slow and inefficient so I bashed together something faster ages ago but never got the time to polish it up for inclusion (it needs things like the various exif tag names and a bit more parsing to be seriously useful). Anyway, glad it works for you (I'm slightly surprised I managed to find it!).

Now ... onto the quality. The [default thumbnail settings](http://picamera.readthedocs.io/en/release-1.12/api_camera.html#picamera.PiCamera.capture) are (64, 48, 35), i.e. produce a thumbnail 64x48 in size with quality 35. Obviously that's very small and low quality so I bumped it up a bit in the gist to 320x180 at quality 50 (which seemed reasonably to me for an actual thumbnail). You can specify any resolution (up to capture resolution) and any quality you like (JPEG quality scales from 0..100) *but there's a catch*...

The Exif data in a JPEG is held inside a single JPEG APP1 block which is limited to 64Kb in size. So all the Exif metadata (timestamp, makernote, camera model, exposure speed, etc.) and the entire thumbnail JPEG itself has to fit within 64Kb. Realistically the Exif metadata is never larger than about 1Kb so that still leaves 63Kb for the thumbnail. When you're dealing with actual thumbnails that's generally fine but if you're wanting to produce thumbnails >1024px wide ... well, that's unlikely to fit in that space (but then at nearly half the horizontal resolution of the original image, I'd hardly call it a thumbnail anymore).

I'm not aware of a way around that limit so if you really need thumbnails that huge I'm afraid you'll have to take the performance hit and scale the original yourself.

Oh, just for reference, the "513" and "514" constants come from the TIFF specification (Exif data is largely in TIFF format). If you have a look right down the end of [TIFF6.pdf](http://www.simplesystems.org/libtiff/document.html) (the last published version of the TIFF spec) you'll find a table of values for all the TIFF tags. If you tweak the gist to dump out all the tags it reads, you'll find lots of interesting stuff in there but tags 513 and 514 are what you want for the thumbnail. Tag 514 specifies the length in bytes of the embedded JPEG thumbnail, and tag 513 points to its offset from the start of the Exif data.

pirquessa, 2017-01-25:
Thank's again for your help and all theses explanations. It's very clear! I just got a new Idea: combining the two solutions:

```
CAPTURE IMAGE ~650ms
EXTRACT THUMBNAIL FROM EXIF AND DISPLAY IT TO USER ~150ms
CONVERT A GOOD QUALITY THUMBNAIL AND DISPLAY IT TO USER ~650ms
```

I can run in two thread the two thumbnail strategy so timing will be:

0 - 800ms : black screen
800ms - 1300ms : First preview with low quality
1300ms - more : High quality preview

It could give the user the impression of a faster process with a progressive rendering of the preview... It sounds to be a good solution :)

pirquessa, 2017-01-26:
It works very well !

First display of the low quality preview arrives after 950ms, the good quality preview arrives after 1400ms. It gives a real impression of speed. :)

Thanks again for your help and your very good lib !

closed 2017-01-26 by pirquessa