waveform80/picamera #360: Different aspect ratio/resolution from same code, on different pi's

emmanuel-contreras, 2017-01-22:
I have 6 raspberry pi's capturing video with the same script (link below). They are all Pi's V3 and camera module V2. The problem is that 2 of the Pi's capture in widescreen and crop the FOV and 4 work as expected. Pi's all have same OS and software as I loaded the same raspbian image on them. 

sensor_mode is specified as 2, and gpu mem set to 256mb. 

https://github.com/emmanuel-contreras/rpi_usb/blob/master/circularBuffer_latest.py

Anything else I should look into to fix this issue? Thanks!

waveform80, 2017-01-23:
Would there happen to be different monitors attached to the Pi's? If so, then because you're not explicitly specifying the capture resolution in the PiCamera constructor (or afterwards), the capture resolution will be defaulting to the resolution of the attached monitor.

Note that specifying the sensor mode does *not* specify the capture resolution; it merely dictates the mode the sensor will be capturing in but the imaging pipeline can still crop and scale to whatever resolution is requested (or implied). I should probably make that clearer in the docs!

emmanuel-contreras, 2017-01-23:
They are all headless now, and I ssh into them. I used a 15" 4:3 monitor to view/align the cameras originally and for the initial setup. Would specifying the resolution (of sensor_mode 2) force the capture of the video to that resolution? Thank you

waveform80, 2017-01-23:
Hmm, if they're headless the picamera is *meant* to default to 720p but maybe something weird is going on there. Anyway, as you've suggested, if you specify the resolution you'll get that as the capture resolution (you can do this in addition to the sensor mode):

```python
from picamera import PiCamera
camera = PiCamera(resolution=(3280, 2464), sensor_mode=2)
```

Or you can just use the `MAX_RESOLUTION` constant which uses the maximum resolution for whatever version of module happens to be attached:

```python
from picamera import PiCamera
camera = PiCamera(resolution=PiCamera.MAX_RESOLUTION, sensor_mode=2)
```

waveform80, 2017-01-23:
Oh, might be worth specifying the framerate too (just to make sure everything's within the bounds of mode 2):

```python
from picamera import PiCamera
camera = PiCamera(resolution=(3280, 2464), framerate=15, sensor_mode=2)
```

emmanuel-contreras, 2017-01-23:
Awesome I will give this a try!

emmanuel-contreras, 2017-02-10:
Quick Update: specifying the resolution and framerate made the videos produced uniform across all pi's. but I had to change the resolution to 1640x1232 for sensor_mode 2 otherwise I got the following error.  "mmal: Too many macroblocks/s requested".

closed 2017-02-10 by emmanuel-contreras