waveform80/picamera #361: Long Exposure Not Working

BootstrapB, 2017-01-22:
I have the latest version of picamera but can't get the long exposure to work. I'm using the recipe 3.7 listed in the docs but if I do print(camera.shutter_speed) it outputs 1705018 and if I time the length of the capture it is only about 2 seconds meaning it isn't capturing for 6 seconds. Any help would be appreciated!

waveform80, 2017-01-23:
Hmm, can't say I can reproduce here. The following version of the low-light recipe is what I tried:

```python
from picamera import PiCamera
from time import sleep, time
from fractions import Fraction

camera = PiCamera(resolution=(1280, 720), framerate=Fraction(1, 6))
camera.shutter_speed = 6000000
camera.iso = 800
sleep(30)
camera.exposure_mode = 'off'
print(camera.shutter_speed)
start = time()
camera.capture('dark.jpg')
print(time() - start)
```

And for me, this outputs:

```
5999884
19.7952840328
```

Indicating that the shutter speed is indeed ~6s (shutter speed is rarely precisely what you set as it's actually a multiple of the line read-out time) and that the capture took ~20s (which is about what I'd expect as it'll involve a couple of mode switches each of which will take at least 6s at that shutter speed, plus the capture time itself).

BootstrapB, 2017-01-23:
Damn, when I ran that exact program it still was only taking about 2 seconds and I tried with 2 different Pi's and cameras. Which model Raspberry Pi and camera are using? I think all my firmware is up to date, but to make sure what OS are you using and which version is your firmware? Also what version of python?

Thanks!

waveform80, 2017-01-23:
That's with a V1 camera module (can't say I've tried long exposures on a V2 yet - they're meant to be able to do 10 seconds but I just adjusted the limits in the code based on that information; it's effectively untested in picamera). The current development Pi is a 3B but most of the recipes (including this one) have been tested on a model B, and a 2B as well. The uname output is:

```
Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
```

Python version won't make any difference (as the library is just a thin wrapper over libmmal and the same code is used for py2 and 3), but for reference it's 3.4.2.

BootstrapB, 2017-01-23:
Turns out our firmware wasn't updated and now it works with the V2 camera. In the docs it says you can update using:

$ sudo apt-get update
$ sudo apt-get upgrade

but previously that wasn't actually updating everything and instead I had to do:

$ sudo apt-get update
$ sudo apt-get dist-upgrade

so you might want to consider changing that in the documentation. Thanks for the help and creating the library!

jaimemarijke, 2018-11-14:
@waveform80 could you go into a bit more detail about the "mode switches" that cause the 6s exposure to take 20s to actually capture?

I've been trying to take 6s exposures, and with only a time.sleep(2), it takes 60s for my capture to complete. I'd like the capture time to be much closer to the exposure time, if possible. 

I tried pre-empting the "mode switching" by setting sensor_mode on initialization, but that didn't make a difference.

Here's my code:
```
from fractions import Fraction
from time import sleep, time

import picamera
import picamera.array

start_time = time()

with picamera.PiCamera(framerate=Fraction(1,6), sensor_mode=2) as camera:
    camera.iso = 100
    camera.shutter_speed = 6 * 1000 * 1000
    camera.exposure_mode = 'off' 
    camera.awb_mode = 'off'

    with picamera.array.PiBayerArray(camera) as stream:
        sleep(2)
        camera.capture(stream, 'jpeg', bayer=True)
        
        print('shutter speed:', camera.shutter_speed)

        # Setting the framerate back above 1 allows camera.close() to actually succeed
        # https://github.com/waveform80/picamera/issues/528
        camera.framerate = 1

print('Elapsed time:', time() - start_time)
```

Output:
```
shutter speed: 5999976
Elapsed time: 51.01670241355896
```

picamera V1.13 with camera module V2.1 on Pi 3 B+