waveform80/picamera #362: Use flash on RPI3

pirquessa, 2017-01-26:
Hi,

I'm trying to add a flash to my RPI3 + Camera V2. I follow the documentation [here](https://github.com/waveform80/picamera/blob/master/docs/recipes2.rst) since [the public one](http://picamera.readthedocs.io/en/release-1.12/recipes2.html?highlight=flash#using-a-flash-with-the-camera) is not up to date for the RPI3 (Thx #325).

Firstly I tried to use the CAMERA_0_LED. But I think there is no more LED on the Camera V2. So I tried to use the P11(BCM17) like in the exemple:

I add the line **_pin@p17 { function = "output"; termination = "pull_down"; }; // CAMERA FLASH_** in the **_pin_config_** section of **_pins_3b2_** and update the **_pin-define@FLASH_0_ENABLE_** of the **_pin_defines_** section to:

```
            pin-define@FLASH_0_ENABLE {
               type = "internal";
               number = <17>;
            };
```
Then compile with: **_dtc -I dts -O dtb dt-blob.dts -o dt-blob.bin_**

I get a lot of warning like:
```
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@default has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p2 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p3 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p5 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p6 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p14 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p15 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p16 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p27 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p40 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p45 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p46 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p47 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p48 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p49 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p50 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p51 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p52 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_config/pin@p53 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_rev1/pin_defines/pin_define@HDMI_CONTROL_ATTACHED has a unit name, but no reg property
...
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@default has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p14 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p15 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p47 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p48 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p49 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p50 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p51 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p52 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_config/pin@p53 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /videocore/pins_cm/pin_defines/pin_define@EMMC_ENABLE has a unit name, but no reg property
```
But there is no error so I apply the file for raspbian with command **_sudo cp dt-blob.bin /boot/_** then reboot.

Try to take some pictures with **_camera.flash_mode = 'on'_**, no success at all. (when I switch my wire from P11(BCM17) to P1(3.3V) i get some light, I think my wiring is good)

Then I simply try to light the LED with some python code (and re wire P11(BCM17) toP15(BCM22)) :
```
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setup(22, GPIO.OUT)
GPIO.output(22, True)
time.sleep(1)
GPIO.output(22, False)
```
It works... So I don't understand why I don't see any light for the flash.

If anyone as an idea...

6by9, 2017-01-27:
The V2 camera doesn't have an LED fitted, but the line still goes up the ribbon cable. On the Pi3 it isn't direct off a GPIO though (it goes through an I2C GPIO expander), so not the best of choices for flash control anyway.

Your blob changes appear correct. Use ```sudo raspi-gpio --get``` to check that your blob has been loaded and gpio 17 is set up as you'd expect as an output.
You should be able to try the torch flash mode to turn the LED using the flash driver - I'm assuming picamera has plumbed that option through.
Are you using Raspbian or NOOBS? It's not the fun with NOOBS having multiple boot partitions?

pirquessa, 2017-01-27:
I'm using Raspbian. I copied the file in the /boot/ partition.

I'll try tonight the `sudo raspi-gpio --get` command and keep you informed. I think it's well mapped, I had an error when I tried to use RPi.GPIO and BCM17.

To test the feature I use the exemple code provided in the doc:
```
with picamera.PiCamera() as camera:
    camera.flash_mode = 'on'
    camera.capture('foo.jpg')
```

Any other options ?

6by9, 2017-01-27:
Checking with http://picamera.readthedocs.io/en/latest/api_camera.html#picamera.PiCamera.flash_mode
```
with picamera.PiCamera() as camera:
    camera.flash_mode = 'torch'
```
should be valid and turn the flash on. Torch mode is typically used either as a torch (remember this code was written for mobile phones), but is sometimes also used for extra illumination during video recording.

pirquessa, 2017-01-27:
Top ! I was wondering if i miss the flash on the led because it was too fast... I'll try something like:

```
with picamera.PiCamera() as camera:
    camera.flash_mode = 'torch'
    time.sleep(1)
    camera.flash_mode = 'off'
```

THX

pirquessa, 2017-01-27:
Hi,

Here is the result of command: `sudo raspi-gpio get`

```
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=0 alt=  func=INPUT
  GPIO 03: level=1 fsel=0 alt=  func=INPUT
  GPIO 04: level=1 fsel=0 alt=  func=INPUT
  GPIO 05: level=1 fsel=0 alt=  func=INPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=0 alt=  func=INPUT
  GPIO 08: level=1 fsel=0 alt=  func=INPUT
  GPIO 09: level=0 fsel=0 alt=  func=INPUT
  GPIO 10: level=0 fsel=0 alt=  func=INPUT
  GPIO 11: level=0 fsel=0 alt=  func=INPUT
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=0 fsel=0 alt=  func=INPUT
  GPIO 15: level=1 fsel=0 alt=  func=INPUT
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=0 alt=  func=INPUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=0 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=0 fsel=4 alt=0 func=PWM0
  GPIO 41: level=0 fsel=4 alt=0 func=PWM1
  GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
  GPIO 43: level=0 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=1 fsel=0 alt=  func=INPUT
  GPIO 45: level=1 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=4 alt=0 func=SDA0
  GPIO 47: level=1 fsel=4 alt=0 func=SCL0
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
```

PIN 17 seems to be an OUTPUT as expected. The code following code produce no results :(
```
with picamera.PiCamera() as camera:
    camera.flash_mode = 'torch'
    time.sleep(1)
    camera.flash_mode = 'off'
```

pirquessa, 2017-01-28:
No more success with FLASH_1_ENABLE :( I really don't understand the problem :(

waveform80, 2017-01-28:
Couple of comments:

* The latest (rendered) version of the docs can be found at http://picamera.readthedocs.io/en/latest/

* You say you're using Raspbian, but did you install it via NOOBS? A simple way to check is to run `mount` and see what device is mounted on `/boot`. If it's `/dev/mmcblk0p5` then you're on NOOBS in which case, as 6by9 suggested, you need to follow the mounting steps for NOOBS (the file doesn't go in `/boot`). If it's `/dev/mmcblk0p1` then you're on plain Raspbian and `dt-blob.bin` should indeed go in `/boot`.

pirquessa, 2017-01-28:
Hi, Thanks again for your answer. 

I'm on a train, I can't test right now but since the result of 'sudo raspi-gpio get' show that gpio17 is an output, I think the file 'dt-blob.bin'  is correctly used, no?

waveform80, 2017-01-28:
If you've run absolutely nothing else since boot-up, yes. But if, for example, you've run a script like the following:

```python
from RPi import GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(17, GPIO.OUT)
```

Then `sudo raspi-gpio get` will output:

```
...
  GPIO 15: level=1 fsel=0 alt=  func=INPUT
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
...
```

Even after the script has terminated as it's left the pin in the OUTPUT state. In other words, from my end, it's difficult to be certain that the device-tree blob is definitely installed correctly without knowing every step that's taken place prior to this.

pirquessa, 2017-01-29:
Hi,

I run the `mount `cmd:
```
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

I confirm, I use raspbian.

pirquessa, 2017-01-29:
I use the file from: `https://github.com/raspberrypi/firmware/raw/master/extra/dt-blob.dts`, after modifications, diff is:

```
pi $ diff dt-blob.dts{.original,}
1162a1163
>             pin@p17 { function = "output";  termination = "pull_down"; }; // Camera flash
1226,1227c1227,1229
<             pin_define@FLASH_0_ENABLE {
<                type = "absent";
---
>             pin-define@FLASH_0_ENABLE {
>                type = "internal";
>                number = <17>;
```

Then compile & copy: `sudo dtc -I dts -O dtb -o /boot/dt-blob.bin dt-blob.dts`, then reboot `sudo reboot`.

Result: nothing :(

waveform80, 2017-01-29:
> Is that really normal to get all the warning I mention on the first post during compilation of dt-blob.dts ?

I don't recall tons of warnings last time I tried this, but that was quite a while ago and I know the devicetree source has changed a lot in that time. I'll try and have a go with this next week.

6by9, 2017-01-29:
I've just tried compiling the blob on my Ubuntu laptop - no errors reported.

pirquessa, 2017-01-29:
I compile it on the RPI with up to date raspbian. Was I supposed to compile it on other file system ? Trying to compile the original file give me the same warnings...

6by9, 2017-01-30:
Just tried on a Pi - no errors.
What does ```dtc -v``` report for you? I've got V 1.3.0 on the Pi, and 1.4.0 on Ubuntu.
Try compiling to the local directory, and then copying the dt-blob.bin file to /boot with sudo. Should be no different, but you never know.

I'll try setting this up fully on a Pi later.

pirquessa, 2017-01-30:
```
pi@pi:~ $ dtc -v
Version: DTC 1.4.1-gbd1c3aca-dirty
```

I was compiling in the local directory. I change lately to gain some time on my failures ;-) It is not different.

waveform80, 2017-01-30:
Okay, looks like something changed between DTC 1.4.0 (from Debian upstream) and DTC 1.4.1 (which looks like a custom rpi build) which causes it to dump tons of warnings upon compilation. I've observed the same thing as both of you: 1.4.1 on the Pi dumps tons of warnings, 1.4.0 on Ubuntu is nice and quiet. I'll add the `-q` switch to the docs to squash the warnings because they aren't errors so there's no sense freaking people out over them.

Now to see if I can get the flash working...

pirquessa, 2017-01-30:
\o/ I'm happy you reproduce my problem ! :)

Thanks again for all your efforts supporting us !!!

[image]

waveform80, 2017-01-30:
Okay, good news and bad. The good news: the flash works (when connected to GPIO17, with the device tree configured as in the documented example). I set `flash_mode` to `'on'`, tried a capture with a V2 module attached and got the characteristic "two blinks" from the attached LED (a blink for measurement, and a blink for capture). Setting `flash_mode` back to `'off'` suppressed the flash. So, I don't think there's any particular issue with the instructions here other than the addition of the `-q` switch to keep DTC quiet.

The bad news: torch mode doesn't seem to do anything (in fact none of the modes except 'on' seem to do anything when capturing). I'd hazard a guess that either this means my environment is too bright for the other modes (although looking at dull grey sky here today, I'm struggling to believe that - it's pretty dingy in the study!), or that only the 'on' flash mode is baked into the firmware, or that something more needs to be done for setting the other modes that I'm not aware of.

Anyway, give the 'on' mode a try and see if you can get a couple of blinks out of your LED. If you can't then it's either a wiring problem (though you've indicated you can light the LED manually via GPIO) or a device tree configuration issue (make sure you've edited both the `pin_defines` and `pin_config` sections, make sure you're in the right block, make sure it's definitely called `dt-blob.bin` in `/boot/` etc).

pirquessa, 2017-01-30:
OMG OMG OMG ! Looking at your response and my diff posted previously:

```
pi $ diff dt-blob.dts{.original,}
1162a1163
>             pin@p17 { function = "output";  termination = "pull_down"; }; // Camera flash
1226,1227c1227,1229
<             pin_define@FLASH_0_ENABLE {
<                type = "absent";
---
>             pin-define@FLASH_0_ENABLE {
>                type = "internal";
>                number = <17>;
```

It didn't ring a bell to see the line `pin_define@FLASH_0_ENABLE` BUT if there is the line, THERE IS a diff... It looks like a `_` became a `-` !!!

I hope it's my problem !!!

waveform80, 2017-01-30:
Ooops - that looks like a bug in the docs. Must admit I just edited the chunk inside the braces. That could well be your issue though. I'll get the docs fixed up to list "pin_define"...

pirquessa, 2017-01-30:
IT WORKSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS

!(THX)[https://media.giphy.com/media/QMkPpxPDYY0fu/giphy.gif]

Thanks a lot for your help !!!

commit 4479ea65 (2017-02-01): Fix #362

Incorrect header to the pin_define section
closed 2017-01-30 by pirquessa