waveform80/picamera #363: seting use_video_port to True causes freeze

robske110, 2017-01-30:
Using the videoport causes picamera to freeze and never return.

Hardware: Raspberry Pi Camera Rev 1.3; Raspberry Pi 3 Model B V1.2
Reproducing code: (If you set the last True to False it works, but that is too slow for me)
```python
import picamera;

class DataWrapper():

    def __init__(self):
        self._rawData = "";
        self._isready = False;

    
    def write(self, data):
        self._rawData += data;
    
    
    def flush(self):
        self._isready = True;

camera = picamera.PiCamera();
cache = DataWrapper();

camera.resolution = (2592, 1944);
camera.rotation = 180;
camera.capture(cache, 'png', True);
```

waveform80, 2017-01-30:
Confirmed, but I'm afraid there's nothing I can do about this: this is a duplicate of #62.

I've had several (intermittent) issues with the PNG codec in the past (which is one reason it's commented out in the test suite; too many cases where it just locks up the camera). It's also not hardware accelerated, so it's pretty slow compared to JPEG (given you're concerned about speed).

If you're using PNG because you don't want any lossy compression at all then just capture an unencoded format like YUV, but frankly I'd suggest you're much better off capturing in JPEG with a high quality level. It'll be *much* faster and the loss of detail is unimportant for the vast majority of applications.

robske110, 2017-01-31:
Well, just tested out the jpeg encoder and wow that's fast. I guess i'll just stick with that. Maybe you should document the problems with png somewhere in the docs. And I'm not actually concerned about the quality and my eyes can't see a difference between YUV and JPEG at 100%.

waveform80, 2017-01-31:
Good stuff - incidentally, if you want high speed processing have a look at the next version of the docs (the ["latest" branch](http://picamera.readthedocs.io/en/latest/) on RTD); I've re-written a lot of stuff in the advanced recipes section including the [rapid capture and processing](http://picamera.readthedocs.io/en/latest/recipes2.html#rapid-capture-and-processing) recipe which has some fancy tricks for doing very rapid JPEG capture (basically splitting up an MJPEG stream).

closed 2017-01-31 by waveform80