waveform80/picamera #364: Question about I2C CSI Communication

AlexDupont93, 2017-02-01:
Hi,
Got a question about the i2c signal from the CSI between the RPi and the MIPI Camera. The instruction of the number of FPS to capture that is sent to the camera, is it MMAL that send the instruction at the begging when we "start_recording" ?

Do you know in what file of PiCamera that send the instruction to the CAM ?

Thanks !

waveform80, 2017-02-01:
I know basically nothing about the specifics of the I2C communication between the camera and the GPU. However, your question indicates some misconceptions about the operation of the camera that the revised [Camera Hardware chapter](http://picamera.readthedocs.io/en/latest/fov.html) in the docs ought to clear up. I'd recommend having a read of that first (I haven't finished revising it yet, but there's already more than enough to clear up the misconception that the camera is idle until recording / capture time).

6by9, 2017-02-02:
It's all done in the GPU blob.

Readout of a line in a mode will always take a fixed amount of time (it varies between modes). Frame rate is controlled by adding extra padding lines to the frame to make the frame take longer to read out.
I'm afraid I can't say which registers it is as I have seen the datasheet under NDA, but you may be able to work it out with an I2C analyser, or from the leaked datasheet that is floating around on the internet.

AlexDupont93, 2017-02-02:
Hi, thanks for the quick reply !
I saw that you made a post on the method to communicate with the I2C GPIO with the Camera Sensor on that article : https://www.raspberrypi.org/forums/viewtopic.php?t=109137

There is no way to send the instruction of the framerate with your method ?

6by9, 2017-02-02:
You are correct that I have not exposed how to set up the frame rate within raspiraw.
Could I draw you attention to the comment at https://github.com/6by9/userland/blob/rawcam/host_applications/linux/apps/raspicam/raspiraw.c#L78 It still stands.

There is a pull request that I need to address for raspiraw (https://github.com/6by9/userland/pull/1) where someone not under NDA has analysed the I2C comms from the GPU and determined flips, exposure time, and analogue gain, and also captured all the modes that the GPU supports. You can use a similar approach if you wish to work out how frame rate control is done, but I can not volunteer the information.

gonzoid, 2017-02-02:
jbeale did some I2C sniffing while running raspistill and raspivid, you can find the capture files on page 2 of this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=47798