waveform80/picamera #368: motion_output callback stop to be called after few seconds

MoogyG, 2017-02-06:
Hi guys,

What an awesome project!

I meet troubles using motion_output. As you can see I start a recording and I want to detect motions (I cut all non necessary code) but after few seconds (from 3 to 30 seconds randomly), "analyse" function stop to be called.

Is it a known issue?

I am using an Rpi1 B(2011) + with a Picam Noir v1 and Python 3.4.2 and Picamera 1.12.

Cheers

```python
import picamera
import time
import signal
import picamera.array

stop_process = False

def signal_handler(signal, frame):
        global stop_process
        print('You pressed Ctrl+C!')
        stop_process = True
signal.signal(signal.SIGINT, signal_handler)

class MyMotionDetector(picamera.array.PiMotionAnalysis):
        
    def analyze(self, a):  
        print("MyMotionDetector analyze...") 

print("Camera init...")

with picamera.PiCamera() as camera:
    
    camera.resolution = (1920,1080)
    camera.framerate = 30
    
    time.sleep(5)
  
    motion_camera = MyMotionDetector(camera)
   
    print("Camera start_recording...")
   
    #stream_file = picamera.PiCameraCircularIO(camera, seconds=20)
    camera.start_recording('/dev/null', format='h264', motion_output=motion_camera )

    while not stop_process:
		
        print('new loop...')

        camera.wait_recording(1)
    
    print("Program closing...")
    
    camera.stop_recording()
    camera.close()

    

```

6by9, 2017-02-08:
What firmware are you running? ```vcgencmd version```?
I'm currently investigating a potential regression in video encode with firmware since 30Dec 2016, but that will only be on your system if you've used rpi-update.

MoogyG, 2017-02-09:
I will make more tests this evening but it seems that there is no problem when I use the resolution 640*480.

Here the results : 
```bash
uname -a
Linux HomeCamPi 4.4.47+ #961 Sun Feb 5 20:11:30 GMT 2017 armv6l GNU/Linux
```
```bash
vcgencmd version
Jan 31 2017 17:44:07 
Copyright (c) 2012 Broadcom
version 8d94fbc3bbdde25c29b18a2572a5c60ee75c56f3 (clean) (release)
```

6by9, 2017-02-09:
Thanks. As your firmware is after Dec 30th I think you're hitting the same as https://github.com/raspberrypi/firmware/issues/728. I'm hoping to have a fix today, so on rpi-update in the next few days.
For reference it is high bit rates and >64kB per encoded frame that hits the issue, so VGA is unlikely to do that.

MoogyG, 2017-02-09:
Good news! I will close the issue after the release, thanks again.

6by9, 2017-02-13:
rpi-update now has what I hope is a complete fix.

MoogyG, 2017-02-14:
I confirm, problem solved on last firmware, thanks.

```bash
uname -a
Linux HomeCamPi 4.4.48+ #964 Mon Feb 13 16:50:25 GMT 2017 armv6l GNU/Linux
```

```bash
vcgencmd version
Feb 13 2017 17:44:42 
Copyright (c) 2012 Broadcom
version a3848f93fbbdb200aa1602b02f925c95c2b138ff (clean) (release)
```

closed 2017-02-14 by MoogyG