waveform80/picamera #371: PiCamera loose its settings after a while

Blindog, 2017-02-12:
I`m getting black pictures if the time between PiCamera configuration and the picture taking is half a second or more.

Example code 

`'''
    Hard to find bug (for me)
    
    if the sleep is small (< 0.3 s) the camera works ok
    if its longer (> 0.5 s) generates a black picture
    
    I guess the camera is loosing its settings
    '''

from picamera import PiCamera
from time import sleep

camera = PiCamera()

camera.iso = 100
camera.exposure_mode = 'off'
camera.shutter_speed = 25000    # choose depending on your light level
camera.awb_mode = 'off'
camera.awb_gains = (0.57,2.08)

sleep(0.5)      # CHANGE THIS VALUE

camera.capture(  "/home/pi/Desktop/image.jpg", format = 'jpeg')

`

I know this code could be easily repaired moving the sleep intruction before, after or supressed. Just to get the idea. 

Camera V2.1
Raspberry Pi 3B

Marcelo

waveform80, 2017-02-12:
Can't say I can reproduce this: I get a very dim JPEG with a heavy blue bias, but that's pretty much what I'd expect from the script above. Bear in mind that you're disabling AGC (setting exposure_mode to "off") immediately after constructing the camera so the gains won't have floated to a reasonable value. There's a reason most picamera scripts pause for a couple of seconds after initializing the camera, which is to let the background AGC loop settle on some reasonable settings.

Blindog, 2017-02-12:
Thanks Dave for your answer (and for your code).

On sleep(0) (or no sleep) I get a nice image. But I get completaly black one from sleep(0.5) or longer.  Not the opposite. Strange.

By the wat. Choose the exposure time and gains as needed. (I'm building a Super 8 scanner. My set is a switched lens camera pointing a negative color film iluminated by a fixed warm (3000 K) led)

Saludos from Buenos AIres.
Marcelo