waveform80/picamera #372: ISO not effected if exposure_mode=off (fixed) at Video/PreviewPort

1kane, 2017-02-12:
I am facing a problem that i cant explain to myself, i read all the docs but im still confused.
Shouldnt the ISO (and therefore gain) value change even when i have set `exposure_mode=off`. (Image and gains changing when its auto)
I noticed that this works if u use the StillPort for capturing images but i really need the VideoPort for my project.
You can reproduce that behavior with ur script from here: https://gist.github.com/waveform80/7a0ca251d1c352b594c18ffb36db2588
set the exposure mode to off (fixed) and then change ISO. the ISO value changes, but the image stays same and also the gain values!!!
Is there something im missing?
the docs says:

> The attribute can be adjusted while previews or recordings are in progress.

but also

> With iso settings other than 0 (auto), the exposure_mode property becomes non-functional.

so is that a bug or is there a workaround for that behavior?

waveform80, 2017-02-12:
Ahh, that's something else that needs correcting in the 1.13 docs: the current docs claim is wrong for exposure_mode 'off'. Here's a better description:

* Setting `exposure_mode` to 'off' fixes the gains and shutter speed entirely. Setting `iso` will make absolutely no difference in this mode (gains will not float and neither will shutter speed, however you can still manually force `shutter_speed` with something other than 0).
* When `iso` is 0 (automatic), and `exposure_mode` is anything other than 'off', gains and shutter speed float.
* When `iso` is non-0, and `exposure_mode` is anything other than 'off', gains are fixed, and shutter speed floats.
* One exception to this appears to be `exposure_mode` 'spotlight' which allows gains to float but limits shutter speed to an extremely low value (<0.2ms)

I'll get the docs fixed up before 1.13.

1kane, 2017-02-13:
Ah okay thank you.
I also found a indirect despriction from `exposure_mode`

> Exposure mode 'off' is special: this disables the camera’s automatic gain control, fixing the values of digital_gain and analog_gain

So for my timelapse project (that should set the ISO automatic higher at too low light , when max shutterSpeed is reached) the only possibility is to set the `exposure_mode='auto'` and then change the ISO setting. I also noticed you should also alter the framerate and there should be some sleep between that. So it looks like that:
```

#exposure=6s and framerate=1/6 ISO=100
#change ISO 
starttime = time.time()
camera.framerate=Fraction(30, 1)
camera.exposure_mode = 'auto'
camera.iso = 200
sleep(1) # is needed or theres no affect
#now lets set the other settings back
camera.framerate=Fraction(1, 6)
camera.shutter_speed = 6000000
camera.exposure_mode = 'off'
print ('ISO change time: %f ms') % (time.time()-starttime)

```
That takes some time, so thats not really best solution for my purpose. It needs at least 1 frame to change and at the next capture another one thts wasted, so u gonna lose 2 frames.

waveform80, 2017-02-13:
Ah, have a read of the [new camera hardware chapter](http://picamera.readthedocs.io/en/latest/fov.html) to get an overview of the operation of the camera. That'll (hopefully) explain why after resetting the sensor (setting the resolution and/or framerate both potentially do this) there typically needs to be a delay to allow the camera to figure out decent gains and/or exposure speeds.

However, you're fixing shutter speed and gains (via "iso") in the script above, so surely no delay is necessary? Unfortunately no: you always need to wait for the next frame for such settings to be applied. As the camera hardware chapter now hopefully makes clear, the camera (once initialized) is constantly in operation: setting properties like "iso" doesn't stop the camera, set the property and restart the camera (resolution and framerate do this because they're affecting the sensor's mode, but everything else doesn't). Rather, setting things like "iso" set the property to be applied when the next frame is captured. Generally, when the camera's running at 30fps and the application is something vaguely interactive that's not a big deal. But if each frame is a hugely long 6 second exposure that does mean after setting everything up you need at least a 6 second delay before the changes will be effective.

I'm afraid there's not much to be gained looking at mmalobj for this sort of thing: the PiCamera class properties like "iso" are very thin wrappers on top of that.

1kane, 2017-02-13:
Thank you very much. I've read the mentioned chapter and of course now i clearly understand why we have to wait for the camera because its always running.
Its just limited by the hardware because it continous reads the next full frame from the sensor and only afterwards changes can be done. If i understand it right than its also limiting that the gains got fixed once u set `exposure_mode='off'`. Okay i guess i have to deal with that.

BTW: i took a look at your new MMAL(Obj) methods and i really like them(some seems usefull for me), i just downloaded the newest repository from here. but i couldnt get your example code running.... arent the new classes ready to try ur exmaples?

waveform80, 2017-02-14:
Ah, I was going to ask for a separate ticket for any mmalobj issues but I see you're ahead of me (with #373 :). I'll have a look at that in a mo (I'm still tweaking mmalobj a bit but the doc examples *should* still work).

commit b23223ad (2017-02-14): Fix #372

Clarify the rather complex relationship between gains, iso, and exposure
mode and explain that exposure_mode "off" is special in both the docs
for exposure_mode and iso. Also duplicated the ISO information from the
camera hardware chapter for the iso property.
closed 2017-02-14 by waveform80