waveform80/picamera #373: testing new MMAL methods

1kane, 2017-02-13:
I really like the new functionality you descriped here : [Docs](http://picamera.readthedocs.io/en/latest/api_mmalobj.html#file-output-jpeg-capture)
I just testet a bit of your expamles (later i wanna redirect previewPort to stream to a local website) , that should connect like:

-  videoPort->Endoder->outputFile

heres the code (simplified) from your example from the docs
https://gist.github.com/1kane/491b6e9646655daa551915844b9266be#file-mmaltest-py

that should output a file that contains 10sec of camera h.264 encoded data. but the file is empty :(
did i miss something`?

waveform80, 2017-02-14:
Huh - interesting! Definitely something going wrong here.

waveform80, 2017-02-14:
Oh, idiot that I am. Was re-factoring the base MMALPythonComponent the other day to have handler methods for all sorts of things that can pass by it and forgot to fix up MMALPythonTarget (unfortunately there's no test suite for Python components in mmalobj yet - and I won't have the time to write one before release, which is part of the reason this API will remain marked "experimental" for 1.13). I'll have this fixed up in a mo.

closed 2017-02-14 by waveform80

1kane, 2017-02-14:
okay thanks im looking forward to 1.13, when its going to be releaed?
seems to have some very usefull nice features!

waveform80, 2017-02-15:
Very shortly is my hope - I was hoping to get 1.13 released before xmas last year, but there was far too much going on (house extension, ill health, excuses excuses) so that date slipped. My plan is to clear down as much of the issues list as reasonably possible this week, to make the release is as comprehensive as possible in the vague hope that people won't notice if I disappear for a few months to hunt down some more profitable work :)

commit a81b0dc6 (2017-02-14): Fix #373

... and tidy up some minor bits in the mmalobj docs