waveform80/picamera #374: Add framerate_range

waveform80, 2017-02-14:
PiCamera's operation when `framerate` is < 1 is confusing: the property returns 0 (which has already led to some support e-mails) and the actual framerate used is a range in these cases. Better to give the user direct control of the framerate and/or framerate ranges.

It would be nice to permit a tuple to be used for this on the `framerate` property, but that's impossible as this is currently permitted as a (deprecated) method of specifying a fractional framerate. Hence this will have to be a separate property (e.g. `framerate_range`). Then question then arises what `framerate` should return when `framerate_range` is set (0?) and what `framerate_range` should return when `framerate` is set (a tuple with the low and high values each set to the framerate?).

Probably need to tweak the night-capture recipe too.

1kane, 2017-02-14:
I am facing a similar frameRate related Problem:
when you set framerate of `mo.MMALCamera.camera.outputs[0] = 0` then the  framerate (or maybe its just the exposure but without framerate limits. ) seems adjusted automatic for the preview (also recording i guess) [ upto 1s - at `exposure_mode='night']`. 
Example: https://gist.github.com/1kane/0d2334f623cecf35b4f77b6525fe92ff#file-autoframerate-py
Disadvantges of that Example is that my preview object gets destoyed after a capture call and im not able to get it back to the the preview port

But there is no chance to get that behavior with the PiCamera Object `PiCamera.camera.framerate=0` seems to do nothig at all - not in the first minute at lowlight.

closed 2017-02-14 by waveform80

waveform80, 2017-02-14:
Yup - framerate 0 (at the MMAL level) is variable frame-rate, in which case the port's framerate range can be specified with MMAL_PARAMETER_FPS_RANGE. This change basically makes it an error to *set* `PiCamera.framerate` to 0, and instead provides `PiCamera.framerate_range` for specifying a (low, high) tuple. When the latter is set, the `framerate` parameter will *return* 0 to indicate a range is in use.

Although that's technically a backwards incompatible change (any script currently setting `PiCamera.framerate` to 0 will break), I'm not terribly worried as currently that functionality does nothing anyway (so one could argue those scripts need changing anyway). It also preserves the current behaviour that when a range is in use, `framerate` returns 0 (in 1.12 and below, setting `framerate` to less than 1 actually sets a variable framerate in one of two ranges to mirror raspistill's behaviour with long shutter speeds).

1kane, 2017-02-14:
Oh thanks!
I'm just wondering that applying setting to `camera._camera.outputs[0].params[mmal.MMAL_PARAMETER_FPS_RANGE]` after `camera=PiCamera.camera` creation doesnt take effect... 
am i missing smthg like commit?

waveform80, 2017-02-15:
Yes I'm afraid it's a lot more complex than that, at least when dealing with the high level PiCamera class. Firstly, yes a commit is needed, but before you can commit the port you need to disable any connection active on that port. So you wind up needing to do something like this (this isn't tested, just off the top of my head):

```python
>>> from picamera import PiCamera, mmal, mmalobj as mo
>>> camera = PiCamera()
>>> camera.resolution = '1024x768'
>>> camera.start_preview(alpha=128)
>>> camera._camera.outputs[0].framerate = 0
>>> r = camera._camera.outputs[0].params[mmal.MMAL_PARAMETER_FPS_RANGE]
>>> r.fps_low = mo.to_rational(1)
>>> r.fps_high = mo.to_rational(10)
>>> camera._camera.outputs[0].params[mmal.MMAL_PARAMETER_FPS_RANGE] = r
>>> camera._preview.renderer.connection.disable()
>>> camera._camera.outputs[0].commit()
>>> camera._preview.renderer.connection.enable()
```

This is more or less what happens every time `PiCamera._configure_camera` is called to reconfigure resolution and framerates.

waveform80, 2017-02-15:
Oh, and I've no idea what setting a positive framerate *and* a framerate range does (I've just assumed when a range is specified framerate has to be 0 because that's what raspistill et al do) - might be something to experiment with, though!

1kane, 2017-02-16:
I got just into a interessting question about the frame/shutter time.
I am trying to do a long exposure(6s) timelapse with taking pictures every 10seconds.

because the camera is always running it keeps taking pictures every 6seconds. but i want it to stop after every picture and resume after 4 seconds of sleep. 

i have tried to stop the PreviewPort with `camera._preview.renderer.connection. enable = False` but that didnt the trick

Is there a way  to "stop" the camera after every capture? and the resume it?
i guess reinitalizing every time would take too much time

waveform80, 2017-02-17:
Stopping the camera means closing the MMALCamera component. However, I don't think would help much: my understanding is that the first frame from the sensor is always corrupt and needs throwing away (which the firmware deals with silently) so when you "resume" the camera you'd have at least a 6 second delay while that occurred.

commit ed6b5568 (2017-02-14): Fix #374

Added framerate_range, some rudimentary tests, and updated the long
exposure recipe