waveform80/picamera #377: Getting back from interrupted script (out of resources)

AdrianAntunez, 2017-02-21:
Hi!

As the picamera has a lot of flexilibity, I have integrated it using boost::python to my C++ program.

In my application I create a PiCamera() instance at the beginning in order to continuously update the awb algorithm. The main problem occurs if the application closes unexpectedly, when I try to run it again I have the following error:

./externalC++App
> mmal: mmal_vc_component_enable: failed to enable compononent : ENOSPC
> Traceback (most recent call last):
>   File "<stdin>", line, in <module>
>   ...
>   ...
> picamera.exc.PiCameraMMALError: Camera component couldn't be enabled: Out of resources (other than memory)

I know that is due to a unreleased camera object in the previous execution (but I couldn't manage it because I got a segmentation error).

Of course, the best solution will be fix all that unexpected endings, but we are in a really early stage so that may occur sometimes (we are working to decrease the amount of such errors).

The main question is, there is a way to check if the camera is busy at the beginning and release it if needed? whether in Python or in command shell, both solutions will work for me.

P.S: I would have tagged that "issue" as a "question" but I didn't find a way to do it.

AdrianAntunez, 2017-02-24:
I didn't came up with a solution to that issue yet.

Does anyone have some workaround? It will be extremely useful to me.

Thanks in advanced 👍

AdrianAntunez, 2017-02-28:
Nothing? :(

waveform80, 2017-02-28:
Even abnormally terminated programs usually result in the camera firmware cleaning stuff up on the GPU side of things. I'd hazard a guess that something in Boost.Python is preventing that happening (there's an interesting note in their "Getting Started" section: "Note that at this time you must not call Py_Finalize() to stop the interpreter. This may be fixed in a future version of boost.python." which *might* have something to do with it).

Overall, I'd suggest if you're using C++ you're better off just using MMAL rather than trying to shoe-horn picamera into your code. If you *really* want to use C++ (presumably for performance reasons) with picamera I'd suggest using it the other way around: write the bulk of your stuff in Python, write the "hot" bits that need optimizing in C/C++, then call them from Python (either by importing them via ctypes or better still by writing them using the [Python-C interface](https://docs.python.org/3/extending/extending.html)).

AdrianAntunez, 2017-03-06:
Hi waveform80,

Thank you so much for your guides in the right direction. In fact I do call `Py_Finalize()` when the code ends up normally because it releases properly the picamera object without errors. I am not sure why it works when in the Boost.Python documentation as you correctly pointed out it says that `Py_Finalize()` shouldn't be called.

As a workaround I forced a `Py_Finalize()` call at the beggining of my code before a Py_Initialize(), by that way, if the interpreter wasn't properly closed, I do force the closing, and thus release the picamera object, before initializing a new one. Otherwise the `Py_Finalize() doesn't take effect.

Thank you so much for you help!

Cheers,

A

closed 2017-03-06 by AdrianAntunez