waveform80/picamera #378: Question: Previewing two videos on the same output

nigelherritt, 2017-02-22:
I'm trying to make an output that splits two video streams along a common horizontal axis and then places them atop and below each other. One video stream is coming from the camera module attached to my pi3B and the other is coming from the network where another pi3B is streaming its picamera video.

I've explored the API references for the PiCamera class and some of the issues raised on github, but am novice with python and raspberry and my attempts thus far haven't been remotely successful.

Essentially, is there a method to preview half of a video stream while displaying it in a certain position (top of screen, or bottom of screen)?

6by9, 2017-02-24:
> Essentially, is there a method to preview half of a video stream while displaying it in a certain position (top of screen, or bottom of screen)?

I don't quite follow you on that bit. Preview half the video stream, ie crop the encoded data? Why send the data if you don't want to display it.

For moving around the preview image, amend your start_preview call to something like
```
camera.start_preview(fullscreen=False,window=(100,200,400,300))
```
100,200 is the x,y offset of the top corner of the preview image. 400,300 is the width,height of the preview image.
Docs are at https://picamera.readthedocs.io/en/latest/api_renderers.html#pirenderer, passed in via https://picamera.readthedocs.io/en/latest/api_camera.html#picamera.PiCamera.start_preview

nigelherritt, 2017-03-11:
Thanks for the response, your comment helped me position the preview image from the camera attached to the pi (let's call it Pi_1). I ended up with the following code in my python script

`camera.start_preview(fullscreen=False,window=(0,0,1024,300),crop=(0,0,1024,300))`

Successfully placing the preview without distortion into the top half of the screen (my camera.resolution was set to the same as the HDMI screen = 1024x600)

Now, I am trying to simultaneously access a stream from my local network and display it in the bottom half of my HDMI display. The server is another pi (let's call it Pi_2) and the stream is RTSP. I have been struggling with Pi_1 (established as the client) to even display the stream through vlc or omxplayer and I suspect it may be my firewall settings. That aside, how may I access this RTSP stream within python and manipulate it such that it previews on the bottom of the HDMI output?