waveform80/picamera #383: I'm lost...in overlay handling

kfeger, 2017-03-02:
Dear All,
I was silly enough to run a apt upgrade of my Raspian-installation today. Before that I had a perfectly running Twitter-App, intended for use at the comming CeBIT in Hannover, Germany.
After the upgrade the following (unchanged) Code
*******************
further above
       AlphaSet = 128
......
	# Logo-Handling für Display
	HDMIimage = Image.new("RGB", (HDMIwidth, HDMIheight))
	#logoNotAus = Image.open('/home/pi/FotoUnten.png')
	logoNotAus = Image.open('/home/pi/SachsenUnten.png')
	logoAugeAuf = Image.open('/home/pi/AugeAuf.png')
	logoAugeZu = Image.open('/home/pi/AugeZu.png')
	HDMIimage.paste(logoNotAus, (0, 750), logoNotAus)
	overlay_renderer = None
	if not overlay_renderer:
272		overlay_renderer = camera.add_overlay(HDMIimage.tobytes(),
		layer=3,
		size=HDMIimage.size,
275		alpha = AlphaSet);
*****************
gives me the following error:
*******************
Traceback (most recent call last):
  File "./tweet.py", line 275, in <module>
    alpha = AlphaSet);
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 892, in add_overlay
    renderer = PiOverlayRenderer(self, source, size, format, **options)
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 420, in __init__
    'unable to determine format from source size')
picamera.exc.PiCameraValueError: unable to determine format from source size
*********************
I assume, the error is actually thrown at line 272 and I have to admitt, I have no clue whatsoever.
Any help? Mr./Mrs. Google didn't give me a hint.
Karl

waveform80, 2017-03-03:
Hmm, something odd going on with the format detection in 1.13 there. Quick fix for now is simply to add the parameter `format='rgb'` to your `add_overlay` call, but is there any chance you could post the image file you're using somewhere? The format detection code is evidently not working properly in this case.

kfeger, 2017-03-03:
Thanks!
I found that same solution today. There seem to be some serious incompatibilities in1.13 to older versions. Now that the old Code is running again, it gives me Memory warnings at another place. But at least, it works for now. CeBIT is save...
My personal Motto 'til then: NOW UPDATES!!

kfeger, 2017-03-03:
Oh, I forgot to answer your question regarding the Image file. I Assume, you mean "logoNotAus ", since the HDMIimage is newly generated as RGB. You find the .png here:
https://dl.dropboxusercontent.com/u/78651193/SachsenUnten.png

kfeger, 2017-03-03:
Since you are bug-hunting, I should probably post the other Problem, too. When the trigegr for a photo ist hit, I Display a Countdown on Screen. This is the Code:
------------------------
		# Countdown Overlay
		Countdown = 3
		while Countdown >= 0 :
			print (Countdown)
			ScreenDraw = Image.new(IFormat, (HDMIwidth, HDMIheight))
			ScreenDrawText = ImageDraw.Draw(ScreenDraw)
			ScreenDrawText.font = ImageFont.truetype(GreetFontpath, 200)
			Numwidth, Numheight = ScreenDrawText.textsize(str(Countdown), font=ScreenDrawText.font)
			ScreenDrawText.text((HDMIwidth - Numwidth,Numheight), (str(Countdown)).decode('utf-8'), (255, 255, 255))
			time.sleep(1)
			Countdown = Countdown - 1
			overlay_renderer.update(ScreenDraw.tobytes())

------------------------
and this is the (new) console Output:
-------------------------
3
2
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1881, in send_buffer
    1
raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1881, in send_buffer
    0
raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1881, in send_buffer
    raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later
Banner aufbringen
An Twitter senden
Fertig
----------------------

kfeger, 2017-03-03:
And finally: Here is what "free -m" says
pi@cebitclick:~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           862        184        677          6         17         62
-/+ buffers/cache:        104        757
Swap:           99          0         99
pi@cebitclick:~ $

So it does not look like a lag of Memory.

poliant, 2017-03-27:
I have same error when calling "update" for the following code

        while True:
            text = time.strftime('%H:%M:%S', time.gmtime())
            img = crossHair.copy()
            draw = ImageDraw.Draw(img)
            draw.font = ImageFont.truetype("/home/pi/font/Calibri.ttf", 50)
            draw.text((10, 10), text, (255, 255, 255))
            overlay.update(img.tostring())
            time.sleep(1)

and also in my case is not a memory problem.
[image]