waveform80/picamera #385: Detect motion, Stop motion detection, full size capture, then resume

FutureSharks, 2017-03-05:
Hi,
I want detect motion, stop or pause motion detection, capture a full size photo in the best quality, then resume motion detection.

```
    def motion_detected():
        print 'motion detected'
        camera.stop_recording()
        camera.resolution = (2592, 1944)
        camera.capture('test.jpeg')
        camera.resolution = (640, 480)
    class MyMotionDetector(PiMotionAnalysis):
        def analyse(self, a):
            a = np.sqrt(
                np.square(a['x'].astype(np.float)) +
                np.square(a['y'].astype(np.float))
            ).clip(0, 255).astype(np.uint8)
            vector_count = (a > MOTION_MAGNITUDE).sum()
            if vector_count > MOTION_VECTORS:
                motion_detected()
    motion_detector = MyMotionDetector(camera)
    while True:
        time.sleep(0.05)
        if camera.recording:
            camera.wait_recording(0.1)
        else:
            print 'starting recording'
            camera.resolution = (640, 480)
            camera.start_recording(os.devnull, format='h264', motion_output=motion_detector)
```

This always hangs though. Any idea how I can get this to work?
Many thanks.

FutureSharks, 2017-03-05:
I've read these issues:
https://github.com/waveform80/picamera/issues/220
https://github.com/waveform80/picamera/issues/125

But didn't help. I'd like to do motion detection at 640x480 or maybe even lower to save on CPU resources but still be able to capture a high quality photo.

pjmelvin, 2017-04-19:
Looks like you are not letting it settle enough. try a sleep(1) somewhere to slow down what you are doing and see if that help.

FutureSharks, 2017-04-21:
Somewhere? That's a bit vague. Where?

I tried this, same problem though:

```
import os
import picamera
import numpy as np
from picamera.array import PiMotionAnalysis
import time

MOTION_MAGNITUDE = 60
MOTION_VECTORS = 20

camera = picamera.PiCamera()
camera.framerate = 24

capturing_photo = False

def motion_detected():
    print('Detected motion!')
    capturing_photo = True
    time.sleep(1)
    camera.stop_recording()
    time.sleep(1)
    camera.resolution = (2592, 1944)
    time.sleep(1)
    camera.capture('full-size.jpeg')
    time.sleep(1)
    capturing_photo = False

class MyMotionDetector(PiMotionAnalysis):
    def analyse(self, a):
        a = np.sqrt(
            np.square(a['x'].astype(np.float)) +
            np.square(a['y'].astype(np.float))
            ).clip(0, 255).astype(np.uint8)
        vector_count = (a > MOTION_MAGNITUDE).sum()
        if vector_count > MOTION_VECTORS:
            motion_detected()

motion_detector = MyMotionDetector(camera)

try:
    while True:
        time.sleep(1)
        if not camera.recording and not capturing_photo:
            print('Starting detection')
            camera.resolution = (1280, 720)
            camera.start_recording(os.devnull, format='h264', motion_output=motion_detector)
        if camera.recording:
            print('wait_recording')
            camera.wait_recording(1)
finally:
    camera.stop_recording()
```

FutureSharks, 2017-05-21:
I got it to work like this:

```python
import picamera
import os
import numpy as np
from picamera.array import PiMotionAnalysis

camera = picamera.PiCamera()

class MotionDetector(PiMotionAnalysis):
    take_photo = False
    def motion_detected(self):
        self.take_photo = True
    def analyse(self, a):
        a = np.sqrt(
            np.square(a['x'].astype(np.float)) +
            np.square(a['y'].astype(np.float))
            ).clip(0, 255).astype(np.uint8)
        vector_count = (a > 60).sum()
        if vector_count > 10:
            self.motion_detected()

try:
    motion_detector = MotionDetector(camera)
    while True:
        if not camera.recording:
            camera.resolution = (1280, 720)
            camera.framerate = 24
            camera.start_recording(os.devnull, format='h264', motion_output=motion_detector)
        if camera.recording:
            camera.wait_recording(0.1)
            if motion_detector.take_photo:
                camera.stop_recording()
                camera.resolution = (2592, 1944)
                camera.capture('full_size.jpeg')
                motion_detector.take_photo = False
finally:
    camera.stop_recording()
```

closed 2017-05-21 by FutureSharks