waveform80/picamera #387: picamera.split_recording drops footage in between splits

nicolashahn, 2017-03-10:
Sample footage of what I'm talking about, the first one happens about 6 seconds in: https://www.youtube.com/watch?v=BxTM42pAGYw

My code: https://github.com/nicolashahn/pi-helmet-cam/blob/master/camera.py

It seems to be dropping about a second or two of footage between splits, but it doesn't happen often, though enough that it happens several times in that clip. Is there any way around this? What's causing it?

paulhothersall, 2017-06-26:
line 97 (or 50 on your 2nd link) 

 camera.wait_recording(interval)

loose that, and set your intrarefresh to the desired length of the splits. This is because its ALREADY waiting for a IFrame and other stuff in the split function. Or roll back to a few version earlier, like 2015 version of picamera..... 

If you really want to poke around, its in lines 825-850 range in https://github.com/waveform80/picamera/blob/master/picamera/encoders.py and I swear there is a logical item or edge case issue there, but i have not found it yet. Yes I can repeat your issue, and its much worse when looking at 2 streams.

In the meantime, my suggested "fix" above should handle it for you

nicolashahn, 2017-12-17:
@paulhothersall your fix worked for me, thank you

closed 2017-12-17 by nicolashahn