waveform80/picamera #389: Web Streaming & Capture simultaneous ?

StormPi, 2017-03-16:
Hello
I tested "Web Streaming" from Picamera: https://picamera.readthedocs.io/en/release-1.13/recipes2.html#web-streaming
And the live works perfectly.

At the same time streaming, is it possible to capture an image every 2 minutes for example?
I will add "server.serve_forever()"; this code:
while True:
camera.wait_recording(120)
camera.capture('foo.jpg', use_video_port=True)

Live works but no images are captured.
Would there be another solution?

Thank

dneunkirch, 2017-03-18:
Hi,
the problem is that `server.serve_forever()` blocks the main thread (no code will be executed after this statement). You have to start the server in an own thread. Maybe helps my motion detection script to solve your problem: https://github.com/dneunkirch/motiondetection/blob/feature/testing/python/motion_detection.py

StormPi, 2017-03-19:
Thank dneunkirch.
I will try to help me with your script.
It's a little more complicated than I thought.

Thank

StormPi, 2017-03-22:
Hi,
I was able to move forward using Thread, thank dneunkirch
`    server = StreamingServer(('0.0.0.0', 8000), StreamingHandler)
    server_thread = Thread(target=server.serve_forever)
    output = StreamingOutput()
    camera.start_recording(output, format='mjpeg')
    try:  
        server_thread.start() 
        while True:     
                time.sleep(120)
                camera.capture('foo.jpg', use_video_port=True)`

The Live works, the file foo.jpg is created but with a weight of 0 ko
Then I get this error message
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end.

Any idea ?
Thank

dneunkirch, 2017-03-23:
Hi,

I don't know your full script, but maybe you have to define a diffrent `splitter_port` for your recording (mjpeg stream) and capture:

`camera.start_recording(output, format='mjpeg', splitter_port=1)`

`camera.capture('foo.jpg', use_video_port=True, splitter_port=2)`

And use:

`camera.wait_recording(120, splitter_port=2)` instead of `time.sleep(120)`

StormPi, 2017-03-23:
Thkank dneunkirch.
I just tried this code:

with picamera.PiCamera() as camera:
    camera.resolution = (640,480)
    camera.framerate = 24
    server = StreamingServer(('0.0.0.0', 8000), StreamingHandler)
    server_thread = Thread(target=server.serve_forever)
    output = StreamingOutput()
    camera.start_recording(output, format='mjpeg', splitter_port=1)
    try:  
        server_thread.start() 
        while True:     
                camera.wait_recording(120, splitter_port=2)
                camera.capture('foo.jpg', use_video_port=True, splitter_port=2)
    finally :
        camera.stop_recording(splitter_port=1)

I get this error:
picamera.exc.PiCameraAlreadyRecording: There is no recording in progress on port 2

I then try this code, although I do not want to save video.h264 file:

with picamera.PiCamera() as camera:
    camera.resolution = (640,480)
    camera.framerate = 24
    server = StreamingServer(('0.0.0.0', 8000), StreamingHandler)
    server_thread = Thread(target=server.serve_forever)
    output = StreamingOutput()
    camera.start_recording(output, format='mjpeg', splitter_port=1)
    camera.start_recording('video.h264', format='h264', splitter_port=2)
    try:  
        server_thread.start() 
        while True:     
                camera.wait_recording(120, splitter_port=2)
                camera.capture('foo.jpg', use_video_port=True, splitter_port=2)

    finally :
        camera.stop_recording(splitter_port=2)
        camera.stop_recording(splitter_port=1)

I get this error:
picamera.exc.PiCameraAlreadyRecording: The camera is already using port 2

Thank.

dneunkirch, 2017-03-23:
You want to capture an image on a splitter_port where is currently a recording. That's my fault. Use time.sleep(120) instead of camera.wait_recording(...) and it should work.

StormPi, 2017-03-23:
No problem.
This is code:

with picamera.PiCamera() as camera:
    camera.resolution = (680,480)
    camera.framerate = 24
    server = StreamingServer(('0.0.0.0', 8000), StreamingHandler)
    server_thread = Thread(target=server.serve_forever)
    output = StreamingOutput()
    camera.start_recording(output, format='mjpeg', splitter_port=1)
    try:  
        server_thread.start() 
        while True:     
                time.sleep(120)
                camera.capture('foo.jpg', use_video_port=True, splitter_port=2)

    finally :
        camera.stop_recording(splitter_port=1)

Unfortunately, I have the same problem.
The Live works, the file foo.jpg is created but with a weight of 0 ko
Then I get this error message
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end.

Thank

StormPi, 2017-03-24:
I forget to ask permission for an image capture during streaming?

import io
import picamera
import logging
import socketserver
import threading
import time
from time import sleep
from threading import Thread
from threading import Condition
from http import server

PAGE="""\
<html>
<head>
<title>picamera MJPEG streaming demo</title>
</head>
<body>
<h1>PiCamera MJPEG Streaming Demo</h1>
<img >
</body>
</html>
"""

class StreamingOutput(object):
    def __init__(self):
        self.frame = None
        self.buffer = io.BytesIO()
        self.condition = Condition()

    def write(self, buf):
        if buf.startswith(b'\xff\xd8'):
            # New frame, copy the existing buffer's content and notify all
            # clients it's available
            self.buffer.truncate()
            with self.condition:
                self.frame = self.buffer.getvalue()
                self.condition.notify_all()
            self.buffer.seek(0)
        return self.buffer.write(buf)

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with output.condition:
                        output.condition.wait()
                        frame = output.frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

with picamera.PiCamera() as camera:
    camera.resolution = (640,480)
    camera.framerate = 24
    server = StreamingServer(('0.0.0.0', 8000), StreamingHandler)
    server_thread = Thread(target=server.serve_forever)
    output = StreamingOutput()
    camera.start_recording(output, format='mjpeg')
    try:  
        server_thread.start() 
        while True:     
                time.sleep(120)
                camera.capture('foo.jpg', use_video_port=True, splitter_port=2)

    finally :
        camera.stop_recording()

Thank