waveform80/picamera #391: Question : Change FPS / Resolution without Close

AlexDupont93, 2017-03-22:
Hi,
I just want to know if we can (after a Stop_recording), change the FPS and Resolution of the camera without closing the camera (with camera.close).

Because sometimes when I do "camera.close" it's stuck on that command and nothing happen...

Thanks a lot guys !

waveform80, 2017-03-23:
Certainly - you can't change resolution or framerate while recording, but otherwise it's fine (actually you can change framerate_delta while recording, but that's another story).

AlexDupont93, 2017-03-24:
Ho, thanks for the quick reply ! But when I stop recording and change resolution its says that there is a problem... do you have a peace of code that you can show to me how to change the resolution without having to close the camera (camera.close) ?

Thanks a lot for the reply I really appreciate it !

Alex

AlexDupont93, 2017-03-24:
And yes I try the framerate_delta and its change the FPS in real time with the MMAL communication with the camera, that's awesome ! But for the resolution it's look like we need to close the camera and I'm trying to look how to do without closing it because randomly I have the bug that it is stuck to the command camera.close and I have to reboot the Pi...

I'm trying to develop a DVR with PiCamera and it's coming pretty good with the PCB that I create for the camera with an Oscioscope that analyze the data that is going to the MIPI communication of the camera!

Thanks !

waveform80, 2017-03-24:
> do you have a peace of code that you can show to me how to change the resolution without having to close the camera (camera.close)?

Sure:

```python
import picamera

camera = picamera.PiCamera()
camera.resolution = '640x480'
camera.start_recording('foo.h264')
camera.wait_recording(5)
camera.stop_recording()
camera.resolution = '720p'
camera.start_recording('bar.h264')
camera.wait_recording(5)
camera.stop_recording()
```

Works fine for me ... if it's not for you I'll need to see the actual code you're using and the error you're getting to debug this any further.

AlexDupont93, 2017-03-24:
Thanks it works! You're the best!

tranqkhue, 2018-09-04:
This doesn't work if I change resolution and then capture still image
`import picamera

camera = picamera.PiCamera()
camera.resolution=(800,600)
camera.start_recording(output='my_video.h264')
camera.wait_recording(10)
camera.stop_recording()
camera.resolution=(3280,2464)
camera.capture('test.png')`

If I wish to capture, then, I have to comment out camera.resolution=(3280,2464)

tranqkhue, 2018-09-04:
This doesn't work if I change resolution and capture still images
```
import picamera

camera = picamera.PiCamera()
camera.resolution=(800,600)
camera.start_recording(output='my_video.h264')
camera.wait_recording(10)
camera.stop_recording()
camera.resolution=(3280,2464)
camera.capture('test.png')
```
If I want to capture still image after recording, I have to comment out camera.resolution=(3280,2464) (or I cannot change resolution)

waveform80, 2018-09-29:
I'm not sure the firmware's PNG encoder will work at that resolution. Unfortunately I can't find a spare V2 module to test with at the moment (I know I've got one *somewhere* but all I can find are spare V1s, so I've ordered another V2!). In the meantime I'd suggest trying lower resolutions, or using JPEG instead (which should be able to handle the full resolution).

tranqkhue, 2018-09-29:
I solved the problem by increasing the memory split of GPU up to 192MB! It seems like PNG format requires larger buffer than JPEG at the same resolution (8MPs for V2 camera). 
Thank you.

waveform80, 2018-09-29:
Ah, good to know - that doesn't surprise me too much though. My understanding is that only JPEG has hardware acceleration (for still image encoding anyway, video has rather more). PNG is thus considerably slower and I doubt has been optimized to the same extent that the JPEG encoder has. Anyway, I'll close this as it seems resolved - do feel free to re-open it if you've more questions!

closed 2018-09-29 by waveform80