waveform80/picamera #393: Issues when testing 1.13 new features (The MMAL Tour in the doc)

MoiCJarod, 2017-03-24:
Hello all and well done for the work on picamera.

This library will be very useful on a project I am working on, especially the new features 1.13 version is adding. I'd like to add some information drawn over a record, so I'm triing to make the examples of "The MMAL Tour" working. The first samples are working well, but when it arrives to what I want, I have some problems to make them work.

For example, I tested the 16.1.11 Python Components example which can be summarized by :

`
from picamera import mmal, mmalobj as mo
from picamera import array

camera = mo.MMALCamera()
preview = mo.MMALRenderer()

class Crosshair(array.PiArrayTransform):
  def transform(self, source, target):
    with source as sdata, target as tdata:
      tdata[...] = sdata
      tdata[240, :, :] = 0xff
      tdata[:, 320, :] = 0xff
    return False

transform = Crosshair()

camera.outputs[0].framesize = (640, 480)
camera.outputs[0].framerate = 30
camera.outputs[0].commit()

transform.connect(camera.outputs[0])
preview.connect(transform)
transform.connection.enable()
preview.connection.enable()
transform.enable()
`

and this code doesn't do anything. What am I doing wrong ? What do I forget ? I tried to eliminate some commands in the previous chapters to keep only the ones that should be necessary for the example ...

16.1.9 example didn't work either, but it seemed to be more complicated that this one.

By the way, I have pointed out some misprint in the doc, where can I spot them ?

Thanks for all :-)

Jean-Rodolphe Letertre

gonzoid, 2017-03-25:
Hello @MoiCJarod, the easiest way to suggest corrections is described here:
https://help.github.com/articles/editing-files-in-another-user-s-repository/

If there are multiple files concerned:

1. do steps 1-6 for the 1st file

2. repeat for the other files but when at step 5, select the same branch ('patch-1' by default)

3. then you can proceed to steps 7-8

This way, all your corrected files will be grouped in a single "Pull Request".

MoiCJarod, 2017-03-25:
Thanks Gonzoid for the tip on how to propose changes to files, but when i try to edit the page with the small pencil, I obtain the message 
"You must be on a branch to make or propose changes to this file"
What should I do to be on a branch ?

gonzoid, 2017-03-26:
I can't reproduce your problem. Alternatively you can try this method:
1. Use the "Fork" button at the top-right to create a copy of this project in your account
2. then you will be able to [edit the files in your repository](https://help.github.com/articles/editing-files-in-your-repository/) (all modifications should be made in the same branch, 'master' by default)
3. finally you can use the "New pull request" button (top-left in your repository).

MoiCJarod, 2017-03-27:
Ok, I eventually was able to do the pull request.

Tried today two example files of the docs :
- mmal_crosshair.py
- mmal_splitter.py

The first one doesn't do anything (waiting at the pause at the end, nothing is showing on the screen), the second is showing me the camera for some seconds with errors printing on console then it crashes.

My camera is a V1.3 one, could it be the reason why it crashes (my rasp is RPi3) ?

Thanks

pirquessa, 2017-03-28:
Hi,

I'm also trying to work with mmal objects to be able to make a nice countdown on the preview. With the [crosshair exemple](https://github.com/waveform80/picamera/blob/master/docs/examples/mmal_crosshair.py), I find that the code hang line 18 and 22. Replacing:

```
            draw.line([
                (port.framesize.width // 2, 0),
                (port.framesize.width // 2, port.framesize.height)],
                fill=(255,), width=1)
            draw.line([
                (0, port.framesize.height // 2),
                (port.framesize.width , port.framesize.height // 2)],
                fill=(255,), width=1)
```

by
```
            draw.line([
                (port.framesize.width // 2, 0),
                (port.framesize.width // 2, port.framesize.height)],
                fill=(255), width=1)
            draw.line([
                (0, port.framesize.height // 2),
                (port.framesize.width , port.framesize.height // 2)],
                fill=(255), width=1)
```

give me some results but it's very laggy, surely due to the error I get in console:
```
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1881, in send_buffer
    raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later
```

But I don't understand this one :( Any idea ?

JimKnowler, 2017-05-15:
I am repeating the same issues + error messages that @pirquessa is reporting.

So far I have tested:
- using apt-get to make sure that python-picamera is upgraded to the latest version
- using apt-get to make sure that all libraries are at the latest version
- running on a PI 2 with Python 2.7.9

After making the fixes suggested by @pirquessa, and while the crosshair example is running, I am seeing a frame of video come through at varying rates between 1 and 3 a second.

To get some insight, I have added debug code to print the state of the pipeline once a second and seeing:

> mo.print_pipeline(preview.inputs[0])

```
 vc.ril.camera [0]                             [0] py.crosshair [0]                             [0] vc.ril.video_render
   encoding    I420                           I420   encoding   I420                           I420      encoding
      buf      2x460800        -->        2x460800     buf      2x460800        -->        2x460800         buf
    bitrate    0bps                           0bps   bitrate    0bps                           0bps       bitrate
     frame     640x480@30fps         640x480@30fps    frame     640x480@30fps         640x480@30fps        frame
```

Interesting that:
- 'bitrate' is reported as 0bps on all components
   - but this is also 0bps when running a working video pipeline with just the camera and preview
- 'buf' is reporting that there are only 2 x I420 buffers on each component
   - compared to 'buf' reporting as 10 x OPQV buffers when running a pipeline with just the camera and preview components

I am going to continue to spend some time investigating, but keen to hear if anyone has any insight.

JimKnowler, 2017-05-16:
The root cause of the issue appears to be commit c4ffce9 (Fix #311).   

I have reverted this commit on this branch and mmal python transforms are working again.  I have tested so far with the mmal_crosshair and mmal_clock_splitter examples.

https://github.com/JimKnowler/picamera/commits/fix_mmal_python_transform

@waveform80 please let me know if there's anything I can do to help this fix make it back to master

DougGore, 2017-06-30:
I can confirm the solution proposed by @JimKnowler fixes the mmalobj issues that cause the pipeline to freeze in the mmalobj examples from the documentation. Thanks Jim!

waveform80, 2017-07-08:
Sadly I haven't got much time to look into this, but I can confirm the issue and the proposed workaround by @JimKnowler. Unfortunately that doesn't help much: read the commit message and you'll see that reverting that commit just breaks #311 again. There's obviously something about MMAL's output buffer handling that I'm not understanding (and hence which goes wrong when the Python-side of the mmalobj components are used). Unfortunately, the pure Python components in `mmalobj` are of extremely niche interest compared to having `capture_continuous` work properly so until I/someone can come up with a "proper" fix things will have to stay as they are.

That  said, if you're playing around with mmalobj, I expect you're more than capable of reverting a commit so I don't feel too guilty ;)

Reinbert, 2017-10-24:
I'm getting the same error when updating an overlay while the preview is running. Unfortunately, on my raspi picamera 1.13 is installed via apt-get and there are no other versions available. How would you recommend to get another version of that library in my case?

waveform80, 2017-10-24:
Apologies I haven't got time to work on this at the moment (it's going to be at least another month before I get free time for picamera again). For anyone wanting to play with older versions:

1. Learn about virtualenvs. There's the [virtualenv user guide](https://virtualenv.pypa.io/en/stable/userguide/) or [pipenv guide](http://docs.python-guide.org/en/latest/dev/virtualenvs/) or probably plenty of others. Anyway, learn it, get comfortable with it.

2. Once you're in a venv you can pip install any version of picamera at all; I never delete old versions from PyPI so everything back to 0.3 (the first version released on PyPI) is still available. Just do `pip install picamera==1.12` (or whatever version you fancy).

3. You can also `git clone` this repository and hack on the code yourself within a venv (there's [development instructions](http://picamera.readthedocs.io/en/release-1.13/development.html) in the documentation).