waveform80/picamera #395: Disable lens shading correction

Blindog, 2017-03-28:
This is a feature request.

**Disable lens shading correction from the ISP**.

This have been discussed here: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=175711 for raspistill. 

By the way I try it, get the code, compile, etc, and I get a different image but not an even image as I get in the rather slow RAW. I have removed the lens from my V2.1 Pi camera.

Thanks

waveform80, 2017-03-28:
Interesting - I should try and keep up with the forums more, but it's difficult to find the time at the mo! Anyway, this should definitely go in.

6by9, 2017-03-28:
> Anyway, this should definitely go in.

If you expose any of the other disables then please put big warnings around it. Disabling random sections of the processing will give very odd effects.

waveform80, 2017-03-28:
Will do! Lens shading can probably go in as a simple True/False property but I'll experiment with the rest and decide whether each should be exposed as a property on PiCamera or not (they'll all go in the MMAL translation, but I figure anyone playing with that knows what they're doing).

Blindog, 2017-03-28:
Thanks. I will be waiting for it. Marcelo

paulhothersall, 2017-06-26:
+1 for this feature AND/OR a relative strength of application.

Using other lenses (M12/CS) and honestly even indoor lighting with the stock lens its not as good as the V1 was on the full sensor size mode 4 (4:3 format to the edge of the frame).

I suspect that lens shader for video was tweaked only on the 1080p and some 720p modes?

scmayo, 2017-08-29:
+1 I too am keen that this feature to disable the ISP blocks as per the original forum link (in first post) makes it's way into the picamera module. Otherwise I may have to hack my camera in C which fills me with horror :-)

I am a researcher experimenting with using the raspberry pi camera for x-ray detection and need to get something like raw bayer/luminance data (at 8 bit) out through the video port. I have promising results using stills port with bayer output but is a bit slow. The video port doesn't do bayer output but the luminance layer of YUV output with all the corrections turned off would be pretty close I'm hoping.

6by9, 2017-09-19:
> I did end up adapting the C code for raspividYUV to switch of the ISP blocks. however even with all the accessible block turned of it appears there is some kind of smoothing still switched on as single photon events are a little smeared in both x and y directions compared to the same in raw Bayer data through the still port.

Surely single photon is only going to activate a single pixel of a single colour, and therefore demosaicing to produce a full resolution image will inherently influence multiple output pixels.
It sounds like your use case _requires_ you to work on the raw Bayer image.

scmayo, 2017-09-19:
Yes I need raw Bayer data - x-ray photons don't notice the colour lenses and none of the normal light corrections are relevant. I was seeing if I could get something close to Bayer output through the video port for improved speed but have had trouble finding out exactly what processing was still being done on video images even when I turned off as much as I could (demosaicing I guess!). 

I'm making do with Bayer output through the stills port now, but would be very interested if raw Bayer was ever available through the video port as this would allow me to collect quicker at higher x-ray flux.  Getting some encouraging results nonetheless. The picamera's small pixel size and low cost makes it interesting for this application (x-ray microscopy).

6by9, 2017-09-19:
@scmayo None of the pipeline is set up to deal with mono sensors or producing mono output which it sounds like what you really want. The effort involved in enabling that is going to be significant.

You could look at [raspiraw](https://github.com/6by9/userland/blob/rawcam/host_applications/linux/apps/raspicam/raspiraw.c). Sorry it's not Python, so you'll have to dig into C again, but it should stream you the raw Bayer frames at full rate (probably faster than you can deal with).
Alternatively the V4L2 driver that has just been [pushed upstream for review](https://www.spinics.net/lists/linux-media/msg121810.html) will allow you easy access to the raw data, and there are likely to be Python libraries for V4L2.

scmayo, 2017-09-19:
Thanks for the pointers -much appreciated - I'll check those things out.