waveform80/picamera #398: Picamera install in virtual environment

MoreSong, 2017-04-14:
I already installed virtual environment.
When I was working on it, I installed picamera by using command:

pip-3.2 install "picamera[array]"

after that, I tried:

python (still in virtualenv)

and then:

import picamera

but at this time, I got an error saying that: "ImportError: No module named "picamera"

And then I tried:

pip freeze

it gave me:

appdirs==1.4.3
numpy==1.12.1
packaging==16.8
pyparsing==2.2.0
six==1.10.0

There is not picamera there!!!

When I log out the virtual environment, and retry python and import picamera, it worked!
But I need to work on the virtual environment, anyone can help me please??

burcuece, 2017-07-21:
Have you managed to solve this problem? I am running into the same issue.

waveform80, 2017-07-22:
You *might* not be using the pip from the virtualenv (typically the pip native to a venv is just called "pip" so there's no need for a version suffix). After activating the venv, ask bash which pip it's using. If it's the one in /usr/bin run `hash -r` to clear out bash's cache of pathnames then try again:

```bash
$ source myvenv/bin/activate
(myvenv) $ which pip
/usr/bin/pip
(myvenv) $ hash -r
(myvenv) $ which pip
/home/pi/myvenv/bin/pip
```

I think modern versions of virtualenv clear the hash table automatically as I haven't seen this in a while, but it's possible you've got an old version if you're still on wheezy.

hupachupa, 2019-03-02:
copy module 'Picamera' (2 folder) from local installation as 'usr/local/lib/python3.5/distr-package' to virtualenv installation '.virtualenvs/py3_for_all/lib/python3.5/site-package'