waveform80/picamera #404: `This system does not appear to be a Raspberry Pi` on openSUSE Leap Pi3

RipCazza, 2017-05-08:
The openSUSE Leap image for the Raspberry Pi fails when installing this package.

```
linux:~ # pip3 install picamera
Collecting picamera
  Using cached picamera-1.13.tar.gz
Installing collected packages: picamera
  Running setup.py install for picamera
    Complete output from command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-dj7gqe9k/picamera/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-cq6oj8c4-record/install-record.txt --single-version-externally-managed --compile:
    running install
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-dj7gqe9k/picamera/setup.py", line 145, in <module>
        main()
      File "/tmp/pip-build-dj7gqe9k/picamera/setup.py", line 140, in main
        cmdclass             = {'install': CustomInstallCommand},
      File "/usr/lib64/python3.4/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib64/python3.4/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib64/python3.4/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-dj7gqe9k/picamera/setup.py", line 111, in run
        raise ValueError('Unable to determine if this system is a Raspberry Pi')
    ValueError: Unable to determine if this system is a Raspberry Pi
    
    ----------------------------------------
Command "/usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-dj7gqe9k/picamera/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-cq6oj8c4-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-dj7gqe9k/picamera
You are using pip version 7.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

6by9, 2017-05-08:
Post the output of ```cat /proc/cpuinfo``` as the error is due to failing to match stuff from there.

Slight alarm bells that you have ```/usr/lib64``` in that backtrace - 64 bit is not officially supported on the Pi. Expect some odd things to happen, and MMAL (which PiCamera relies on) is unlikely to work as there is a void* lurking in one of the IPC structures.

carmenbianca, 2017-05-08:
@6by9 You're correct that picamera doesn't run on a 64-bit OS. That is tangentially related to the issue though. The issue is that the module doesn't install.

Whether it's useful to install a module that will not run... I don't know for sure. I'd wager yes. RTD requires picamera installed to generate docs, and therefore has an exception in `setup.py`. I'd personally do away with the check in `setup.py` altogether, though.

6by9, 2017-05-08:
Up to @waveform80 as to whether he wants to support OpenSUSE as an OS.

And you haven't provided the requested information of what is in ```cat /proc/cpuinfo```

carmenbianca, 2017-05-08:
@6by9 I'm not @RipCazza

6by9, 2017-05-08:
@carmenbianca Oops, sorry!
I wasn't expecting anyone else to be creating PRs in the same area on such a newly logged issue.

carmenbianca, 2017-05-08:
@6by9 To be fair, @RipCazza did ask me to take a look at this for him.  I don't
have access to his Pi, though.

sanduhrs, 2017-07-04:
Having the exact same issue on a pi3 with arch linux:
```
linux:~ # pip install picamera
Collecting picamera
  Using cached picamera-1.13.tar.gz
Installing collected packages: picamera
  Running setup.py install for picamera ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8vno1szp/picamera/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5wsr82__-record/install-record.txt --single-version-externally-managed --compile:
    running install
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-8vno1szp/picamera/setup.py", line 145, in <module>
        main()
      File "/tmp/pip-build-8vno1szp/picamera/setup.py", line 140, in main
        cmdclass             = {'install': CustomInstallCommand},
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-8vno1szp/picamera/setup.py", line 111, in run
        raise ValueError('Unable to determine if this system is a Raspberry Pi')
    ValueError: Unable to determine if this system is a Raspberry Pi
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8vno1szp/picamera/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5wsr82__-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-8vno1szp/picamera/
```
The requested information:
```
linux:~ # cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

sanduhrs, 2017-07-04:
I forced skipping the test by setting the environment variable:
`export READTHEDOCS=True`

But system is still 64-bit so not working.

LizardM4, 2017-09-27:
Development is not necessarily done on the Raspberry Pi directly. It would be useful to at least have the package install on other systems, to have e.g. code completion enabled.

PR #405 (2018-09-29): Remove custom install routine
Fixes #404.

Not all Pi operating systems report a 'Hardware' line in /proc/cpuinfo.

This would allow the package to be installed on non-Pi systems.  This
isn't really an issue, though: People installing this on non-Pi systems
would simply run into runtime errors.

As I understand it, the setup script should contain as little logic as
possible.  But I guess this is one of those things where people have
opinions one way or the other.

And to quote from 06c38bf:

"And this is why you try not to do special cases in your setup.py, kids,
because there's no end once yo [... cut]
commit c1450158 (2018-09-29): Remove custom install routine

Fixes #404.

Not all Pi operating systems report a 'Hardware' line in /proc/cpuinfo.

This would allow the package to be installed on non-Pi systems.  This
isn't really an issue, though: People installing this on non-Pi systems
would simply run into runtime errors.

As I understand it, the setup script should contain as little logic as
possible.  But I guess this is one of those things where people have
opinions one way or the other.

And to quote from the commit message that added this section:

"And this is why you try not to do special cases in your setup.py,  [... cut]
closed 2018-09-29 by waveform80