waveform80/picamera #406: Include subsecond Exif information in pictures

ltog, 2017-05-10:
I'm interested in using the `capture_continuous()` method to capture street sceneries while driving. Later on I'd like to geotag the pictures using a GPX track. Since the camera v2 is able to capture more than one picture per second, there may be cases where multiple Exif tags `DateTimeOriginal` show the same capture time. Because of this geotag software will geotag different pictures to the exact same coordinates, which is annoying when showing the capture locations on a map.

The Exif standard contains tag such as `SubSecTimeOriginal` to store subsecond values that could be used to geotag pictures more precisely. However, it seems that these tags are not written to picture by default.

Could these tags be added to picamera or is there a workaround to get these tags written?

BTW: I read https://github.com/waveform80/picamera/blob/974540f38793c79c69405948e2fffafb3a1025dd/picamera/camera.py#L1774-L1778 but assume this can only be used for Exif tags that don't change over time.

6by9, 2017-05-10:
The firmware is missing the mapping for the string ```IFD0.SubSecTimeOriginal``` to tag number 0x9291. I'll look at adding it.
Most people aren't moving fast enough to make a huge difference within 1 second without motion blur or rolling shutter effects, but there probably are enough to make it worthwhile finding the 5 mins to add it.

ltog, 2017-05-10:
@6by9 : Thank you, I'd be happy I you could have a look!

6by9, 2017-05-11:
I'd looked in the wrong place. EXIF.SubSecTime, EXIF.SubSecTimeOriginal, and EXIF.SubSecTimeDigitized are al supported at the codec level (as correctly documented in the PiCamera docs). I'd looked at code that was automatically adding metadata based on the camera parameters.

@waveform80 would be better placed to answer whether you can update exif tags between continuous captures.

waveform80, 2017-05-11:
Erm ... good question ... I'm pretty sure it should be possible to modify the `exif_tags` dict during iterations of `capture_continuous` (I've had a cursory glance at the stuff down in `encoders.py` which seems to support this assertion). I'd suggest just trying it and let me know if it doesn't work!

ltog, 2017-05-25:
@waveform80 : I set the values of `exif_tags` like this:

    camera.exif_tags['EXIF.SubSecTime'] = datetime.datetime.now().microsecond)[:-3].zfill(3)

This seems to work fine. A complete script can be found at https://github.com/ltog/mapillary_utils/blob/master/raspi/autocapture.py .

Is this what you meant?

Thank you for your hints!