waveform80/picamera #408: Compatible cameras

pirquessa, 2017-05-15:
Hi,

Thanks again for this awesome lib !

I was wondering if there is any other camera compatible with this lib that would be able to take better pictures. Maybe real raspberry pi camera equipped with a lens ? I currently use the official camera V2.

Does someone know if this better camera exists ?

Thanks :)

kif, 2017-06-11:
Hi,

The V1 exists with M12 or CS mounts. I own a CS-mounted one and costs hardly more than a normal camera. You can then add a "good" CS lens.
The V1 work pretty well. 

I built a M12 mount for the V2 but there are multiple issues: 
* threading manually in M12x0.5 is tricky, 
* the sensor does some additional correction which would require additional flat-field corrections 
* M12 lenses, while cheap are usually not that great: there are plenty of geometric correction to apply as well.

Changing the sensor is another story !

Cheers,

pirquessa, 2017-06-11:
Thank you for this answer :)

Anyone else ?

julesrenaud, 2017-06-12:
@kif I'm also working with M12 lenses on the V2. I'm curious, did you manage to figure out the corrections that need to be applied to counter the presets on the sensor?

kif, 2017-06-12:
On Mon, 12 Jun 2017 07:20:17 -0700
Jules <notifications@github.com> wrote:

> @kif I'm also working with M12 lenses on the V2. I'm curious, did you manage to figure out the corrections that need to be applied to counter the presets on the sensor?

This is apparently related to the back-focal-length i.e. to the
distance from the sensor to the back of the sensor. Unfortunately it is
likely to be specific to each individual lense. 

I have another lens, so I could test if the flatfields obtained are the
same but this will require some additionnal work.

My lens is 2.8mm with F/2.8 it shows much geometryc distortion.
I measured a set of images with various exposure time in front of the
back-light of a flat-scanner which is supposed to give an uniform white
image of size ~ A5. The distance from the camera to the light is a
dozen of centimeters.

The methodology I plan to apply:
* Read the image
* convert from RGB to linear intensity space (correct back gamma compression)
* average over rings centered on the image
* scale the result so that the mean is 1

Re-do for all image acquired and check consistency (it should be)

I will send you the images obtained once I got them.

Cheers,

Jerome

kif, 2017-06-12:
Here is the demo: I averaged the images taken with exposure time from 1000µs to 9000µs. Above, the camera shows saturation.
https://github.com/kif/imagizer/blob/qt/timelaps/Flatfields.ipynb
The result is decent. I am not sure about the centring of the lens on the sensor: it is hand-threaten in 3D printed PLA. 

By the way, does anybody know/understand what the exposure_mode means ? (like night/veryslow/...)

Cheers,

pirquessa, 2019-01-05:
Hey, There are many months I start this subject (>18 months), did you see any new device with better quality ? I don't :(

pirquessa, 2020-04-30:
https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/ here it is !!!! <3

@waveform80 will it be supported by your awsome lib ? :) :) :)

6by9, 2020-04-30:
I did check it ages ago, and memory says that picamera checks the resolution of the sensor via camera_info, same as raspistill does.
So yes it will support it, and should default to sensible resolutions too.

pirquessa, 2020-04-30:
EXCELLENT ! Can't wait to test it !

closed 2020-04-30 by pirquessa