waveform80/picamera #411: picamera.exc.PiCameraError: Unable to determine default GPIO LED pin for RPi revision 3 and camera num 1

dcrispgit, 2017-05-22:
I think I have found a bug in the camera.py file in version 1.13.  
I think I have the solution to the bug BUT you will need to confirm. 
I have a Compute Module with a Pi3 with Dual Cameras. 
When I try and address camera 1  ( myCamera = PiCamera(1) )  I get the error: 

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 345, in __init__
    }[(GPIO.RPI_REVISION, camera_num)]
KeyError: (3, 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test2.py", line 4, in <module>
    camera = PiCamera(1)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 350, in __init__
    GPIO.RPI_REVISION, camera_num))
picamera.exc.PiCameraError: Unable to determine default GPIO LED pin for RPi revision 3 and camera num 1

I CHANGED the following line in the file : 
(0, 1): 30, # compute module (default for cam 1)
to
(3, 1): 30, # compute module (default for cam 1)
I could probably have added a new line of code for that but i was trying to work out what was goign on so only managaed a quick and dirty fix. 

The Camera now correctly detects and was accessible in my code.

Goldesel23, 2017-09-06:
I got the same error with a compute module V3 with dual cameras. Your suggestion resolved my problem thanks. 

When I analyze the RPI_REVISION it outputs the RPI3 model instead of compute module. The updates of raspbian and the firmware updates did not solve that, but this did. 

Thank you!

waveform80, 2017-09-07:
Unfortunately I don't have a CM3 to play with, but it definitely looks like this needs updating (I can probably nick some code from GPIO Zero to do "proper" detection of the CM3).

MrPepkins, 2019-02-06:
I had the same error too! Compute Module 3 with 2 PiCameras (NoIR 2.1) attached, fortunately your solution worked for me!! Thanks

6by9, 2019-02-06:
The docs at https://picamera.readthedocs.io/en/latest/api_camera.html#picamera do say

> The led_pin parameter can be used to specify the GPIO pin which should be used to control the camera’s LED via the led attribute. If this is not specified, it should default to the correct value for your Pi platform. You should only need to specify this parameter if you are using a custom DeviceTree blob **(this is only typical on the Compute Module platform)**.

You are using a custom dt-blob.bin (as the CM doesn't enable the cameras by default) and are on a Compute Module, therefore you should really be specifying led_pin to the constructor.

RPI.GPIO seems to be returning 3 for any unknown revision, partly as it is now supposedly a revision of the GPIO header (ie 26pin vs 40pin). The CM1 is special cased to return 0, but not the CM3 or CM3+.

Conversely the list is missing many of the more recent Pi variants, but on many of those the LED is controlled via an GPIO that isn't accessible via RPi.GPIO.

pranav083, 2020-12-24:
The same solution works for me on rpi4 compute module also. But unable to use `DISP0` instead using `DISP1` and both the camera on `CAM1` and `CAM0` are working fine. also need to connect both the jumper on `J6` on the IO Board.

skippy2626, 2023-08-16:
I'm also using a rpi4 CM (attached to a stereopi, if that matters), and this solution worked for me. For anyone curious, you can find camera.py on your pi board at "/lib/python3/dist-packages/picamera"