waveform80/picamera #414: Picamera and Arducam Multiplexer

lefsky, 2017-06-03:
I want to work with 8 Pi cameras simultaneously, preferably using picamera. I have been planning to use 2 RPis with 2 Arducam Multiplexers (which allows you to connect 4 cameras and access the sequentially). 

From what I know of the picamera software, it seems that I would need to close and open each camera when I switched between them (cameras are selected by altering GPIO settings). Is this conclusion correct? 

I suppose that my other option is the use the Pi Zero, although that will require communicating with 12 RPis  rather than 2, which I'd prefer. Will I use lose any functionality with the Zero? 

M

johnsethsalazar, 2019-01-31:
Hello, lefsky I'm experiencing a similar problem that you have if you could help. I would appreciate it very much thank you.

Here's the link of the problem I'm experiencing:
https://stackoverflow.com/q/54417601/9482697

6by9, 2019-02-04:
These multiplexer boards are not made by Raspberry Pi, and are frequently a pain in the neck (at least to me). There is no official support for them from Pi Towers, and Arducam will be unable to assist on the software side as it is all running on the GPU (closed source).

The mux should switch whilst streaming (if not then complain to Arducam), but:
- The camera stack is not designed to expect the camera module to suddenly change without the GPU knowing. You are very likely to get odd results as AE and AWB have to adjust when you switch.
- you may get corrupt frames during the switch, as it is not synchronised to the CSI2 data at all.

In @johnsethsalazar 's case, what you want to do is relatively trivial in that you can set up the multiplexer before enabling the camera. The bigger question to ask is what to do if the second PIR is triggered whilst the first is recording - you can not call start_recording twice to try and record both cameras simultaneously.