waveform80/picamera #419: Error in Example 4.8 Unencoded video capture

stdsmile, 2017-06-29:
When trying to execute the Code Example 4.8 in the [docs](http://picamera.readthedocs.io/en/release-1.13/recipes2.html).
i get:

Traceback (most recent call last):
  File "Example48.py", line 41, in <module>
    camera.add_overlay(memoryview(box), size=(160, 90), layer=3, alpha=64)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 892, in add_overlay
    renderer = PiOverlayRenderer(self, source, size, format, **options)
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 413, in __init__
    source_len = mo.buffer_bytes(source)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 466, in buffer_bytes
    return m.itemsize * reduce(mul, m.shape)
UnboundLocalError: local variable 'm' referenced before assignment

when starting with python3 

Traceback (most recent call last):
  File "Example48.py", line 41, in <module>
    camera.add_overlay(memoryview(box), size=(160, 90), layer=3, alpha=64)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 892, in add_overlay
    renderer = PiOverlayRenderer(self, source, size, format, **options)
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 413, in __init__
    source_len = mo.buffer_bytes(source)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 466, in buffer_bytes
    return m.itemsize * reduce(mul, m.shape)
UnboundLocalError: local variable 'm' referenced before assignment

Greetings

stdsmile, 2017-06-29:
In mmalobj.py in line 466 findet man:

> Given an object which implements the :ref:`buffer protocol <bufferobjects>`, this function returns the size of the object in bytes.  The object can be multi-dimensional or include items larger than byte-size.

`def buffer_bytes(buf):`

    if not isinstance(buf, memoryview):
        m = memoryview(buf)
    return m.itemsize * reduce(mul, m.shape)

jjsanderson, 2017-08-06:
I've just stumbled across the same issue, in the same example code. Pity - it looks like a useful example!