waveform80/picamera #423: UnboundLocalError local variable 'm' referenced before assignment - overlay

dreasGh, 2017-07-05:
I, i have an issue when i'm trying to run the following example: https://picamera.readthedocs.io/en/release-1.13/recipes1.html#overlaying-images-on-the-preview

I'm using python 3.4

Traceback (most recent call last):
  File "picamera.python", line 18, in <module>
    o = camera.add_overlay(memoryview(a), layer=3, alpha=64)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 892, in add_overlay
    renderer = PiOverlayRenderer(self, source, size, format, **options)
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 413, in __init__
    source_len = mo.buffer_bytes(source)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 466, in buffer_bytes
    return m.itemsize * reduce(mul, m.shape)
UnboundLocalError: local variable 'm' referenced before assignment

tuyenld, 2020-12-10:
This was fixed on https://github.com/waveform80/picamera/commit/96b8a0983194dfa7e7d2fd19ab4317961ba1ea44
In case you are installed picamera through `apt-get install python3-picamera`. Here is the **dirty and dangerous** way to hot-fix this issue:
```
# Find picamera loction
$ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera
>>> picamera.__file__
'/usr/lib/python3/dist-packages/picamera/__init__.py'
>>> exit()

# Apply patch
vi /usr/lib/python3/dist-packages/picamera/mmalobj.py

-        m = memoryview(buf)
-    return m.itemsize * reduce(mul, m.shape)
+        buf = memoryview(buf)
+    return buf.itemsize * reduce(mul, buf.shape)
```

esologic, 2020-12-28:
What is the timeline for releasing this fix onto pypi?