waveform80/picamera #426: autorun picamera script in rc.local doesn't actually produce video file

caseyanderson, 2017-07-18:
I have the following code which I can run normally to record video, using `python3 button_led_video.py`, however when I put it into /etc/rc.local the camera no longer records video. Here is the python3 script:

```python
#!/usr/bin/python3
# press and hold to record video

from gpiozero import Button
from gpiozero import LED
from picamera import PiCamera
from datetime import datetime
from time import sleep

# setup led and button
button = Button(4)
led = LED(17)

# recording flag
is_recording = 0

try:

    while True:
        if (button.value == True) and (is_recording == 0):

            camera = PiCamera() # there has to be a better way to do this
            sleep(2)
            camera.start_preview()
            print("CALIBRATING...")
            sleep(2)
            print("READY TO RECORD!")
            print()
            print("RECORDING")
            x = datetime.now().strftime('%Y-%m-%d-%H-%m-%s')
            camera.start_recording('video-' + x + '.h264')
            is_recording = 1
            led.on()

        elif (button.value == False) and (is_recording == 1):
            print("DONE RECORDING!")
            camera.stop_recording()
            camera.stop_preview()
            led.off()
            camera.close()
            is_recording = 0

except KeyboardInterrupt:
    print("INTERRUPTED!")
    button.close()
    camera.close()
    led.close()
```

Here is how I have it in `/etc/rc.local`: `sudo python3 /path/to/file.py &`

When I make `file.py` executable and add the above line to `/etc/rc.local`, following a reboot (when I would expect the script to run on its own) I am able to get the LED to light up but nothing happens with the camera (no new video files appear after pressing, holding, and then releasing the button).

I am totally stuck on this, as it works outside of /etc/rc.local reliably. Any idea what I am doing incorrectly here?

waveform80, 2017-07-18:
It is working, but you haven't considered that the environment that your script is running under is different. For example, when you're testing your script you're logged in as "pi" (presumably) and your current directory is "/home/pi" (presumably). So when the script runs, and you press the button, it starts recording the video under `/home/pi/video-blah-blah.h264` and the file is owned by "pi".

However, the `/etc/rc.local` script is executed as "root" on boot-up (hence you don't need "sudo" in there; it'll be run as root anyway). Moreover, the current directory will probably either be root's home-dir ("/root") or the root of the filesystem ("/") so any videos recorded will be there, and will be owned by "root". I'd bet good money if you do something like "ls /" or "ls /root" you'll find your recorded videos. If you want to delete or move them, you'll need to do so as root (with "sudo") as root will own the files and no other user will be allowed to remove them (although they're probably universally readable).

If you want your scripts to appear in the regular location just specify a full path for `start_recording`. I'd also recommend running your script as the "pi" user to avoid the complications of root-owned files, so in `/etc/rc.local` you'd want something like:

    su -c "python3 /path/to/file.py" pi &

caseyanderson, 2017-07-18:
oh wow, you were right...`ls /` was where all of the videos were going. i made the changes you suggested and am now able to run my script on boot and record video. thanks so much. ill update my question on the rpi stackexchange to include your answer.

for realz, you just made my day. thank you!

closed 2017-07-18 by caseyanderson