waveform80/picamera #427: KeyError when closing capture_continuous

hiramf, 2017-07-28:
I've been running into an MMAL out of resources error for a capture_continuous instance of PiCamera. It works the first time, but when I try to run again, I get the MMAL error. After reading up on the documentation, I realized I should have a `camera.close()` line. Even though it works, the program throws an error:
```
Exception ignored in : <generator object capture_continuous at 0x6b05c918>
Traceback (most recent call last):
    File /home/pi/miniconda3/envs/py34/lib/python3.4/site-packages/picamera/camera.py", line 1713, in capture_continuous
    del self._encoders[splitter_port]
KeyError: 0
```

When I change the `splitter_port` parameter, it changes the `KeyError` to that port number.

Here are the relevant lines from my program:
```
# initialize camera
camera = PiCamera()
camera.resolution = (640,480)
camera.framerate = 16
rawCapture = PiRGBArray(camera, size = (640,480)
[...]
for f in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
    frame = f.array
[...]
if key == ord("q"):
    camera.close()
    break
```

vmanikes, 2018-03-27:
Did you fix it, I am also getting the same error

davesdere, 2018-05-26:
I have the same issue. I do have a WebRTC server running though. It uses the camera. I think we can only use the feed of the camera once.

waveform80, 2018-09-29:
> Additionally, camera.close() only fixes the problem when I'm using a GUI. If I'm going headless and use Ctrl-C or Ctrl-Z to exit the program, I can't start the program again without rebooting.

I should note that Ctrl-Z does *not* quit programs; it merely pushes them to the background jobs of the shell, so the camera will remain "owned" by that process. Ctrl-C *should* quit programs though (unless the signal is being captured / suppressed). I'd have to see more of the source to figure that out though.

> I have the same issue. I do have a WebRTC server running though. It uses the camera. I think we can only use the feed of the camera once.

Precisely, as your update notes only one process on the system can have the camera open at any given time. Other processes attempting to open the camera will receive the MMAL out of resources error.

I'll close this for now, as I think it's just a case of background processes continuing to use the camera, but do feel free to re-open it if you have further information!

closed 2018-09-29 by waveform80