waveform80/picamera #431: PiCameraMMALError after running few days

toannguyencnu, 2017-08-07:
Error after working few days.
Dear all,
I wrote the following code to record the video in 3 seconds and capture images while recording. The code works fine, I can capture and record video. However, after two days, the code has some bugs. I tried to clear the buffer but no luck. 
Could you please help me to over this problem.
Thank you!

The code here:
import time,datetime,sys
from threading import Thread,RLock
import signal
import picamera
import os,shutil
import glob
import logging
import io
from picamera import bcm_host, mmal, mmalobj as mo
from picamera.array import PiRGBArray
import cv2
active_path = '/home/pi/data/active'
backup_path = '/home/pi/data/backup'
temple_path = '/home/pi/data/temple'

_client_id1 = sys.argv[1]
_debug = True
seriNo = sys.argv[2]
client_id = _client_id1 + "-" + seriNo
totalSize = 0
count_n = 1
while True:
    if (count_n > 50):
        os.system("sudo vcdbg reloc")
        count_n = 0
    else:
        count_n = count_n+1
    with picamera.PiCamera() as camera:
        time.sleep(0.1)
        camera.preview_fullscreen = False
        camera.preview_window = (100, 20, 640, 480)
        camera.resolution = (1920,1080)
        camera.framerate = 30

        stream = picamera.PiCameraCircularIO(camera, seconds = 3)
        print '-'*60
        testFile1 = os.path.exists(backup_path)
        camera.start_preview()
        # setup circular buffer
        video_name_temple = temple_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.h264')
        video_name_active = active_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.h264')
        video_name_backup = backup_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.h264')
        print 'Video: ', video_name_temple
        camera.start_recording(stream,format='h264',splitter_port=1)
        camera.split_recording(stream)#, splitter_port=1)
        for i in range(3):
            camera.annotate_text = client_id + "\n" + time.strftime('%Y-%m-%d %H:%M:%S')
            image_name_temple = temple_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.jpeg')
            image_name_active = active_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.jpeg')
            image_name_backup = backup_path +'/' +client_id+'-'+time.strftime('%Y%m%d-%H%M%S'+'.jpeg')
            #filename.append(image_name)
            time.sleep(0.8)
            camera.capture(image_name_temple,  splitter_port=0)

            testTempleImage = os.path.exists(image_name_temple)
            testActiveImage = os.path.exists(image_name_active)
            testBackupImage = os.path.exists(image_name_backup)
            
            print 'Image: ', image_name_temple
            if testTempleImage: ## and testActiveImage and testBackupImage:
                try:
				# Copy data to other folder
                    totalSize = totalSize + os.path.getsize(image_name_temple)
                    shutil.copy2(image_name_temple,image_name_active)
                    shutil.copy2(image_name_temple,image_name_backup)
                except:
                    pass
        # Stop recording
        camera.stop_recording(splitter_port=1)
        # Write video data
        with io.open(video_name_temple, 'wb') as output:
            for frame in stream.frames:
                if frame.frame_type == picamera.PiVideoFrameType.sps_header:
                    stream.seek(frame.position)
                    break
            while True:
                buf = stream.read1()
                if not buf:
                    break
                output.write(buf)
        totalSize = totalSize + os.path.getsize(video_name_temple)
        print 'Total size: ', totalSize
        testTempleVideo = os.path.exists(video_name_temple)
        testActiveVideo = os.path.exists(video_name_active)
        testBackupVideo = os.path.exists(video_name_backup)
        if testTempleVideo: # and testActiveVideo and testBackupVideo:
            try:
                shutil.copy2(video_name_temple,video_name_active)
                shutil.copy2(video_name_temple,video_name_backup)
            except:
                pass
        camera.stop_preview()
        if totalSize > 100000000: # size >100MB
            try:
                shutil.rmtree(temple_path, ignore_errors=True)
                os.mkdir(temple_path)
                print '--------------Clean Temple--------------'
                totalSize = 0
            except:
                pass
        camera.remove_overlay
        camera.close()
        stream.seek(0)
        stream.truncate(0)
        stream.clear()
        stream.close()
        B = mo.MMALQueue(camera)
        B.close
        B.seek(0)
		
        camera1 = mo.MMALCamera()
        preview1 = mo.MMALRenderer()
        encoder1 = mo.MMALImageEncoder()
        encoder2 = mo.MMALVideoEncoder()
        camera1.outputs[2].disable()
		preview1.disconnect()
		preview1.inputs[0].disable()
		encoder1.disable()
		encoder1.outputs[0].disable()
		encoder1.inputs[0].disable()
		encoder2.disable()
		encoder2.outputs[0].disable()
		encoder2.inputs[0].disable()
 
    
Errors after runing 2 days:

Video:  /home/pi/data/temple/KH0101AO-0002-20170802-103848.h264
mmal: mmal_vc_port_enable: failed to enable port vc.ril.image_encode:out:0(JPEG): ENOMEM
mmal: mmal_port_enable: failed to enable port vc.ril.image_encode:out:0(JPEG)(0x8d1000) (ENOMEM)
Traceback (most recent call last):
  File "dfst_capture_v2.py", line 59, in <module>
    camera.capture(image_name_temple, use_video_port=True)#, splitter_port=2)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1418, in capture
    encoder.start(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 1126, in start
    super(PiCookedOneImageEncoder, self).start(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 377, in start
    self.output_port.enable(self._callback)
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 1249, in enable
    prefix="Unable to enable port %s" % self.name)
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Unable to enable port vc.ril.image_encode:out:0: Out of memory

Thank you for your kindly helps!

openedhardware, 2017-10-20:
Same issue for me. Would be +1

briankrupp, 2018-01-25:
We use a circular buffer as well in our project and have experienced memory leaks monitoring for motion. We isolated the code so that we removed anything that was unique in our implementation and we still encounter an out of memory error which occurs after several hours (anywhere from 5-8 hours):

`
class MotionDetector(picamera.array.PiMotionAnalysis):
    def analyze(self, a):
        return

streamMaxSeconds = 30

# Set resolutions up for capture and detection
captureWidth = 1024
captureHeight = 768

# Setup Camera
camera = PiCamera()
camera.resolution = (captureWidth, captureHeight)
camera.rotation = 270
camera.framerate = 30

# Setup Circular stream to monitor for motion
stream = picamera.PiCameraCircularIO(camera, seconds=streamMaxSeconds)

camera.start_recording(
    stream, format='h264',
    motion_output=MotionDetector(camera)
)
print("Started monitoring for motion")
while True:
    time.sleep(1)`