waveform80/picamera #432: capture timeout

z-afshar, 2017-08-11:
Hello,
I just bought 5 megapixel OV5647 camera, and enable it through raspi-config
the camera works well with raspistill and raspivid utilities and takes images and videos well,
but when I use the picamera module to capture pictures it shows me an error after capture command:

>>> from time import sleep
>>> from picamera import PiCamera
>>> 
>>> camera = PiCamera()
>>> camera.resolution = (1024, 768)
>>> camera.start_preview()
<picamera.renderers.PiPreviewRenderer object at 0x76458f30>
>>> # Camera warm-up time
... sleep(2)
>>> camera.capture('foo.jpg')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/picamera/camera.py", line 1423, in capture
    'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end

I'm stuck on this step with picamera module, please help, thanks

simonek0007, 2017-09-17:
Hi, 
did you solve the problem? I have the same situation.

waveform80, 2017-09-18:
@z-afshar @simonek0007 what environment are you running from? The script works happily for me from the Python command line, and from IDLE but I've recently seen something similar on SO and I'm wondering if you two are using something else which is doing something odd with threading?

simonek0007, 2017-09-19:
Hi, I formatted the SD card and install Raspbian from the beginning. I have Raspberry PI Desktop version. Below you can find the steps which I made: 
1. sudo -apt-get update 
2. sudo apt-get install python-picamera python3-picamera 
3. sudo apt-get upgrade 
4. uname -a Linux Raspberrypi 4.9.41-v7+ #1023 
5. sudo reboot 
6. run the script (python3 script) for capture the image after capture the image i can only restart the python shell. The led in camera is lighting all the time.
7. In python 3.5.3 Shell choose "Debug" and "Debugger" then I run the script and run it step by step. till error appear: <frozen importlib._bootstrap>:182:_get_module_lock() KeyError:'picamera' After that the image is capture and I can see it on the LCD, but the LED is still lighting. I need to press CTRL+F6 to reset the python shell, then LED is turn off. In Python Shell window I got error message: Timed out waiting for capture to end. picxamera.exc.PiCameraRunTimeError. The image is saved on the SD, but when I try to open error message appear: There is no data.
Please see pictures of error below.
[image]
[image]
[image]
[image]

simonek0007, 2017-09-30:
Any help?

Patataman, 2017-11-10:
I have mostly the same problem with the Camera V2 on Python2.7. raspistill works perfectly, but when I try the simplest code with PiCamera, it's always a timeout.

sunilkumarit, 2018-06-23:
Thank U
It is helpful

RognanCr, 2018-10-29:
Ok, update. 

So i've bought official picamera V2 and it's working how it supposed to.

I've read about people having the same issue. No solution works. Most people don't specify the type of the camera so it is hard to tell which type does not work (in my case "RPi Camera (D) 5Mpix REV 2.1 (Waveshare 11297)"). 

Of Course it only matters if you want to use still_port through this python interfaces.