waveform80/picamera #437: Problems with split_recording in different thread

squidpickles, 2017-08-31:
I am running a multithreaded application that does camera capture into a buffer.
I have a thread that is just looping on `wait_recording()`, so that I can get exceptions.
When I try to split the recording from another thread, the `wait_recording()` loop ends up taking zero time to wait, and spins as fast as possible. Meanwhile, no frames are sent to the new buffer.
If I split the recording from inside the `wait_recording()` thread, though, it seems to work fine.
No exceptions are raised or anything.

Here's an example that reproduces the problem:
```python
import threading
import time

import picamera

BUFFER_SIZE = 2 * 1024 * 1024

class Camera(threading.Thread):
    def __init__(self):
        super().__init__()
        self.camera = picamera.PiCamera()
        buf = picamera.CircularIO(BUFFER_SIZE)
        self.running = True
        self.camera.framerate = 10
        self.camera.start_recording(buf, format='h264',
                                    bitrate=1500000, intra_period=10,
                                    sei=True)

    def split(self):
        buf = picamera.CircularIO(BUFFER_SIZE)
        self.camera.split_recording(buf)

    def run(self):
        while self.running:
            print(" - wait_recording")
            self.camera.wait_recording(1)

    def stop(self):
        self.running = False
        self.camera.stop_recording()
        self.camera.close()

camera = Camera()
camera.start()
print("Waiting 5 seconds")
time.sleep(5)
print("Splitting recording")
camera.split()
print("Waiting 5 seconds")
time.sleep(5)
camera.stop()
```

juhaj, 2020-07-31:
@squidpickles did you ever figure this out? I seem to have ran into the same issue. It seems something inside `picamera` (or even deeper underneath it) is not thread-safe.
I also see the same error if I place the `split_recording` call inside the `analyze()` method of a subclass of `picamera.array.PiRGBAnalysis` - but I suspect the underlying problem is the same as that method also runs in a thread different from the main interpreter thread.

squidpickles, 2020-07-31:
@juhaj I never did figure it out. Ended up ensuring all camera commands happened on the same thread, which worked around the problem.

juhaj, 2020-08-05:
I've done the same now. I ended up having them all in the main thread, but I guess that's strictly not necessary and any thread would do as long as only one of them accesses the camera.
It also seems that only methods that interact with the hardware are problematic because stuff like the `analyze` method of `PiRGBAnalysis` work just fine even if they're in another thread provided they don't touch the hardware and operate on the provided numpy array only.

squidpickles, 2020-08-05:
Yeah, I ended up setting up a `ThreadPoolExecutor` with a single worker to handle all of the hardware stuff; method calls to my own camera abstraction are wrapped in a decorator that dumps them into the executor's queue. Everything else remains undecorated, and runs in the caller's thread. A little bit complicated, but it works well.