waveform80/picamera #439: Incorrect padding in SPS header

LizardM4, 2017-09-09:
Problem
--------
When I compare the raw h264 data extracted with picamera to the raw h264 data after muxing the stream into an mp4, I see that in all SPS headers a block of 6 bytes has been shifted left (and padded with zeroes). This 1-byte shift causes some applications to fail to play the video (e.g. Telegram when sending the media, QuickTime on OSX).

How to reproduce
------------------
**Record a h264 stream (any recipe would work):**
```
import picamera
camera = picamera.PiCamera()
camera.resolution = (640, 480)
camera.start_recording('test.h264')
camera.wait_recording(1)
camera.stop_recording()
```
**Mux and demux the stream:**
```
$ ffmpeg -f h264 -i test.h264 -c:v copy -an -f mp4 test.mp4
$ ffmpeg -i test.mp4 -c:v copy -an bsf:v h264_mp4toannexb test2.h264
```
**Binary compare the two streams:**
[image]

Notes
------
The problem is not in demuxing process, the same difference can be spotted if comparing the `mdat` section of the mp4. Replacing the `mdat` data content in a well formed mp4 (such as the ones produced by ffmpeg or VLC) with the original h264 data causes the stream not to be playable.

ffplay and VLC seem to do just fine, but if ffmpeg can "fix" that piece of stream this is not surprising.

I printed the various frame sizes from `camera.frame` every time a complete frame was written to the stream, and those 6 bytes are exactly the last 6 bytes of the SPS header. If another key frame is requested, it will end with exactly that sequence and it will also be shifted by 1 in the muxing proces.

If I record at a different resolution, e.g. 1080p, I see the same issue. The first 3 bytes of the shifted block are actually even the same (`01 28 EE`).

----

I do not know how h264 works internally, but I found no indication anywhere that the h264 stream should be changed while muxing. *I suspect the SPS header coming from the camera contains a small padding error.*

LizardM4, 2017-09-09:
Sorry for bothering. I figured that the change is due to the fact that the AVC streams use a different format for specifying NAL units in the h264 stream, in shorts those `00 00 00 01` markers are replaced by the length of the NAL unit (and there may be something else still going on).

The information though was in a different ISO standard, namely section 5.2.3 of ISO 14496-15.

For future reference, [this question](https://stackoverflow.com/questions/5544696/parsing-h264-in-mdat-mp4) on SO mentions the gist of the problem.

closed 2017-09-09 by LizardM4

waveform80, 2017-09-09:
> I am trying to write a custom output for picamera that outputs the fully muxed mp4 directly, and I encountered this weird difference. I do not know how h264 works internally, but I found no indication anywhere that the h264 stream should be changed while muxing. I suspect the SPS header coming from the camera contains a small padding error.

Ah! This is something I've been wanting to get around to for literally years now, and have never found the time (damn having to earn money! ;). If you get something working, and fancy contributing it to the project, I'm more than happy to do integration bits (docs, tests, clean-up, whatever) though I'd certainly understand if this was a private project.

LizardM4, 2017-09-10:
Actually if I get it to work, I'd be more than happy to contribute. As of now, I figured out the mp4 atoms that are needed (ISO 14496-14) and managed to mux the stream thanks to [pymp4](https://github.com/beardypig/pymp4). I hope I can work out the remaining steps too!

LizardM4, 2017-09-10:
Hey @waveform80, it seems I got it to work; I need to re-arrange this a bit, then I'll open a PR.