waveform80/picamera #443: v1.3 capturing empty frame (black screen) on Pi 2

abhisheksoni27, 2017-09-16:
v1.10 works fine, though. :thinking:

waveform80, 2017-09-16:
Erm, any reason you can't use 1.10 (or later) then? I mean, 1.3 was 3 years ago!

Or did you mean 1.13? If the latter, could you provide a code sample so I can try and reproduce the issue?

abhisheksoni27, 2017-09-17:
Typo, Sir. :laughing: 

I meant 1.13.

Here's the code I am trying to run:

```
import cv2
from picamera.array import PiRGBArray
from picamera import PiCamera
import time

camera = PiCamera()
rawCapture = PiRGBArray(camera)

time.sleep(1)

camera.capture(rawCapture, format="rgb")
image = rawCapture.array

cv2.imwrite('Image', image)
```

rwb27, 2017-09-21:
I think I've come across this recently too.  For me, the captured images are black only if I am using full resolution (Pi 3, camera module v2).  If I decrease the resolution to half of that, it works fine.  Capturing to png images on disk or rgb format arrays at full resolution fails, but capturing to jpeg files is OK.

I have not tried changing the ``picamera`` version, but I have tried increasing the GPU memory split, which is often cited as causing this problem.  I've tried both 256 and 512Mb with no avail.

Oddly, if I use a PiBayerArray, I can reconstruct a full quality raw 16-bit image with no trouble!

GeneGIGA, 2018-02-02:
I have also run into a similar issue. Here is my behavior...tested with both current version 1.13 and 1.10.

To get the behavior...
1. set resolution to 640 x 480
2. exposure_mode set to 'off'
Status: captures images just fine in both still and video ports
3. set resolution to 1024 x 768
Status: black frame appears from still port...video port works fine...just not the still port
4. exposure_mode set to 'auto'
5. exposure_mode set to 'off'
Status: captures images just fine in both still and video ports

Also tried adjusting shutter_speed, etc. to see if the frame was just dark...but no, there is no information in the image at all as far as I can see...guessing that as rwb27 described...it is there but not converting correctly

rwb27, 2018-02-05:
Just to check - are you using jpeg captures, or something lossless like format="rgb" or saving as a png?

rwb27, 2018-02-05:
I could capture fine at up to half of the raw resolution, as long as I was happy with a JPEG...

GeneGIGA, 2018-02-05:
Am using jpg captures to test. 
Produces black frames at low and high resolutions...640x480, 1024x768...etc.

rajatsaxena, 2018-02-14:
I am able to capture jpg at half of the raw resolution as well,