waveform80/picamera #445: make PiBayerArray usable without camera hardware

rwb27, 2017-09-21:
``PiBayerArray`` is a nice way to recover the raw image data from a JPEG image saved with the pi camera.  I use it when I need access to the raw image data on-the-fly, but it would be nice if my post-acquisition analysis code could use it too.  When running on a Pi it's fine, as the ``picamera`` library is installed - but it would be great if I could install it on my PC as well.

I currently do this using a hacked version of array.py, with a stub version of the mmal objects module.

Is this something that could be usefully achieved by a slight reorganisation of code in PiCamera (e.g. refactoring to move the PiBayerArray code to its own module), and if so would a pull request make sense?  If not, I'll probably bodge something together for my own purposes, perhaps with a script that will pull changes from picamera to make it more easily maintainable.

When writing code that talks to hardware I've often found it convenient to allow it to install on platforms where that hardware is never present, so it can run as a stub to allow testing/debugging/analysis using the bits of the code that don't actually depend on the hardware.  However, it may be that this is not done, by choice or convention, for this project and in any case it would amount to more work than it's probably worth in this case...

thatneat, 2018-12-04:
Hey @rwb27, my team just released a fork which accomplishes this. Check out https://github.com/OsmoSystems/picamraw/

rwb27, 2018-12-10:
thanks - that's helpful :)