waveform80/picamera #451: Some help/guidance much appreciated

upyesp, 2017-11-02:
Hi,

Great work on the Python library for the camera.  It has helped me SO much.

Would really appreciate some help on the things I'm looking for below.  Have searched forums and the PiCamera documentation with no luck. My questions aren't issues, but I guess if they aren't possible they might be feature requests?

For my project, on the video 'preview' and 'stream', I...

o need to reduce the colour depth to 2 colours e.g. black & white (not grey scale)
o need to swap those two colours for other colours e.g. black to blue and white to yellow

Is it possible to achieve this with the Python library currently?

Thanks in advance.

6by9, 2017-11-02:
Duplicates forum post https://www.raspberrypi.org/forums/viewtopic.php?f=43&p=1229796

waveform80, 2018-09-29:
Looks like this was answered pretty comprehensively in the forums.

closed 2018-09-29 by waveform80