waveform80/picamera #452: Question: how much processing has been done on Bayer output data?

mklein9, 2017-11-02:
I am using picamera in an instrumentation application using a monochromatic laser source and without lens.  To get the least processed image possible I am using the bayer=True option to capture_continuous() and the PiArrayOutput() renderer. I have disabled all processing options I can find.

I have read the 1.13 docs (which are great, BTW) but can't really find an explicit indication of where the Bayer output is pulled out of the pipeline.  For example, the Bayer array must come out from some point in the sequence of GPU actions listed in section 6.1.6 ("Background processes").  Does anyone know which actions have been done on the array before it appears as the Bayer array?

As one experiment, I set camera.hflip = True, and see that the Bayer array comes out horizontally flipped.  So at least transposing is done on the raw data before sending the Bayer array out.  It would be really nice if the Bayer came out at least before lens shading was done.

6by9, 2017-11-03:
The Bayer images are as received by the SoC from the sensor.
The only parameters that therefore affect them are mode selection, exposure time, analogue gain, and H & V flips (NOT transpose, which is specifically reflecting in the line x=y).

This is all in the docs - http://picamera.readthedocs.io/en/latest/fov.html#background-processes

mklein9, 2017-11-03:
OK thank you.  I might suggest to clarify in the docs that it is in fact the Bayer data after mode, exposure, a-gain and flips that travels to the GPU, and the bayer option to the capture() functions intercepts it at that point.

Thank you again, the overall description was very valuable and highly useful.  And the library is working very well in my application.

closed 2017-11-03 by mklein9