waveform80/picamera #455: Bug: Changes to preview resolution cause preview to freeze when recording started.

atomiclama, 2017-11-24:
The following code works fine, so all hardware and software is assumed to be ok.
Raspberry Pi Zero W
rasbian all uptodate 
live preview on composite out.

I want to record at a HD resolution but I require a different resolution on the composite out as it is hooked upto a 4:3 analogue TV. If I insert the commented out code then I get 5 sec of correct preview on the composite output but as soon as the recording is started the preview freezes.

Will check HDMI when I get access to one. 

`camera = PiCamera()
camera.resolution = (1280, 720)

camera.start_preview()

#camera.preview.resolution = (720, 576)

sleep(5)

timestr = strftime("%Y%m%d-%H%M%S")

camera.start_recording(timestr+'.h264', format='h264', quality=23)
camera.wait_recording(60)
camera.stop_recording()`

6by9, 2017-11-28:
(a) Your use case makes little sense as your preview resolution is a 4:3 aspect ratio when the recording is 16:9. That would mean you aren't seeing the same thing as you're recording.

(b) It's an inherent limitation of the camera subsystem that recording and preview must be at the same resolution. Setting ```resolution``` and ```preview.resolution``` differently only really makes sense when taking JPEG or YUV/RGB capture

(b) It's an inherent limitation of the camera subsystem that recording and preview must be at the same resolution. Setting ```resolution``` and ```preview.resolution``` differently only really makes sense when taking JPEG or YUV/RGB captures using the ```capture``` or ```capture_continuous``` methods.

(c) The display hardware will rescale almost anything to anything. It'll know that the output is 720x576 for composite and will scale/crop all display resourse accordingly. You can adjust exactly where in the output scene your camera resource goes with the ```window``` property of the renderer http://picamera.readthedocs.io/en/latest/api_renderers.html#picamera.PiRenderer.window

atomiclama, 2017-11-28:
@6by9 Thanks for taking the time to respond. 

(a) Have to disagree here, I don't need to see all that I'm recording just the centre part of the display is good enough. I still think this is a valid use case. 

(b)(c) fair enough if this is a limitation of the subsystem

(d) Discovered that the display hardware is re scaling to the 720x576 through trial and error.  I'm currently using 
`camera.start_preview(crop=(160,0,960,720))` to crop out the centre section and allow the display HW to scale to full screen output. This is working ok.

Do you have any pointers for frame rate settings when used with composite output. Do I need to set to 25fps to match with normal PAL 25/50 interlacing? 

Cheers

atomiclama, 2017-11-28:
As stated above 

>  It's an inherent limitation of the camera subsystem that recording and preview must be at the same resolution

closed 2017-11-28 by atomiclama

6by9, 2017-11-28:
> Do you have any pointers for frame rate settings when used with composite output. Do I need to set to 25fps to match with normal PAL 25/50 interlacing?

The display framework at vsync will take whichever frame was last presented to it and then display that in the next frame. If your source and sink rates differ then every so often you will get one frame dropped or duplicated to compensate. If that is an issue to you then run the source (camera) at the same rate (or a multiple thereof) as the sink.
Mismatched rates are not an issue to the system, but may have display artifacts that you will object to.

atomiclama, 2017-11-28:
Thanks, that makes sense.