waveform80/picamera #456: How to synchronize two camera‘s framerate

deepzf, 2017-12-16:
I'm going to start a project using several pi zero w with picamera.But I don't kown how to let them capture at the same time Accurately .I hame done some tests and several different settings,but still about a unsynchronized frame at intervals of twenty frames.
Is there any Solution to solve this problem?

LoudMurmur, 2018-01-15:
to make a stereo camera you need a "sync" pin on each camera, this is the only way to synchronize to cameras.

I do not believe the raspberry camera have such features.

dethesut, 2018-01-29:
hoihj

soswow, 2018-07-23:
I will just leave it here: As I understand there is a way to get digital signal when frame is taken https://www.raspberrypi.org/forums/viewtopic.php?t=190314 But I am still not sure how to use it to ensure two frames taken at the same time.

waveform80, 2018-09-29:
Actually, I should add something to the docs about the GPIO pulses ...

maykef, 2018-11-05:
A bit late for this comment but I have managed to sync both cameras on a compute module by using multiprocessing. Here is an example:
[image]

soswow, 2018-11-06:
@maykef Well Done, Mate! I was sure it is possible to do with compute module ;-) Not sure how, since dont have one. Maybe not easy.  I believe people mostly want to find a way to do it with separate PIs.

deepzf, 2018-11-06:
my point is how to sync cameras on different pi boards

maykef, 2018-11-06:
Not sure multiprocessing can run different processes in different machines, I’ll have a quick look into it.

maykef, 2018-11-06:
@deepzf  This might help you:
http://www.micsymposium.org/mics_2017_proceedings/docs/MICS_2017_paper_4.pdf

soswow, 2018-11-06:
@maykef  Do you have any more to say about what you did? Maybe you have a blog post or smth?

maykef, 2018-11-06:
@soswow 
This is my entry from 2 years ago:
https://www.instructables.com/id/A-Raspberry-Pi-Multispectral-Camera/
This is from this year:
https://publiclab.org/questions/maykef/10-12-2018/cheap-multispectral-camera 
What is it that you want to do?

soswow, 2018-11-06:
@maykef I guess I am interested in this like everyone else here - to do stereo CV

maykef, 2018-11-06:
Thought so.
Someone already developed this:
http://stereopi.com 
All I can tell is that one thing is to trigger the camera in still mode (rolling shutter, overheads and the like), and another one very different is to sync frames for stereo vision. I have seen few projects were people managed to sync several cameras, so it's got to be possible.

soswow, 2018-11-06:
I've seen this one and even pre-ordered already, I believe.

maykef, 2018-11-06:
So you are on your way then!

maykef, 2018-11-21:
Hi @ninja-asa, 

First of all, well done for that. Now, as I understood, the reason why you ditched the CM was because of the connection between the cameras and the CM board, is that correct? 
When you say you connect 2 RPi 3, one as master, the other as slave, is that done through network, or you just simply connected both GPIOs to initialise the video capture?
It sounds interesting, and I can actually reply it with a bunch of RPis I've got, and compare performance with the compute module.

soswow, 2018-11-21:
@ninja-asa Interesting. I assume you are doing for same reason (stereo CV) I wonder what is the threshold in terms of time difference for stereo CV to be still practical?

ninja-asa, 2018-11-21:
@maykef 
1) Actually I still use the CM, the amount of GPIO can be really handy . The syncing I proposed was just a work-around that for me works best to optimize a process of making the stereo-calibration. Until then, I must have spent countless hours verifying connections trying to understand why the cameras wouldn't be detected by the compute module: was it the flat cable between CM and Camera adapter? was it the flat cable between the adapter and the camera? was any of the  jumpers connecting camera -related GPIO broken? This happened to me quite often and connecting/disconnecting I would end up breaking those black supports on the Compute Module Board. This frustration is way smaller with just using two RPi3.

2) No network, only hard-wired GPIO, the simplest and most straightforward approach possible. (But really important is to have the RPi share GND otherwise it is possible that the HIGH of the slave "is not clearly noticed". (Actually I use network but is only to remotely access the master RPi )

3) By the way, thank you for sharing the StereoPi project. Really looking forward to its launch.

@soswow  I guess it really depends on the context of the stereo you intend to do, the expected/target working distances and which maximum velocity you expect in your view and of course your hardware

maykef, 2018-11-22:
Hi @ninja-asa,

I've got the suspicion that GPIO connections won't allow for camera shots sync, I actually did this long time ago already. I'll give another try and report here.

realizator, 2018-11-30:
Guys, StereoPi is a project of our team.
As I know there is no hardware sync (at sensors levels) used in Compute Module based solutions. But our experiments with depth map and livestream stereoscopic video to users (Oculus Go) shows that synchronization issues are undetectable.
If you plan to use depth map for drone races with 100 miles per hour speed - of course it is better to use hardware sync, and also use sensors with global shutter. But in this case, it is better to look for some industrial-level solutions. StereoPi is designed for beginners who study OpenCV and stereoscopic vision, and for this scenario synchronization implemented in stock Raspbian kernel now is perfect.