waveform80/picamera #459: MMALPythonPort does not support MMAL_ENCODING_H264 and MMAL_ENCODING_MJPEG

barney-NG, 2018-01-14:
Hi Dave,

first of all: thank you for this excellent peace of work! 

By writing a h264 decoder I tried to use MMALPythonSource as simple method to create a source stream for the decoder pipeline. 
I think I found 3 different errors (or may be I just used it wrong...)

1. Both video formats are not included in the supported_formats property of the MMALPythonPort class. (I added them manually in my installation to dig deeper in the stuff)
2. The copy_from() helper raises an error. (may be I added the formats not correctly) source.outputs[0].connect(decoder.inputs[0]) raises an error when source.outputs[0].format is not set explicitly.
3.  source.connection.enable() is not supported.
4. MMALPythonSource.send_buffer() is raising: PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')

Thank yo for any help!

Here my code I used for testing:
[mmal_h264_decoder_example.py.txt](https://github.com/waveform80/picamera/files/1629392/mmal_h264_decoder_example.py.txt)

barney-NG, 2018-01-16:
Ok -- I debugged this topic a bit and did a "minimal invasive" patch.
(I don't feel comfortable enough with Python to add patches by myself...)
The streaming input still overruns the decoder output capacity :-/

_Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/pi/picamera/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
  File "/home/pi/picamera/picamera/mmalobj.py", line 1931, in send_buffer
    super(MMALPortPool, self).send_buffer(port, block, timeout)
  File "/home/pi/picamera/picamera/mmalobj.py", line 1881, in send_buffer
    raise PiCameraMMALError(mmal.MMAL_EAGAIN, 'no buffers available')
picamera.exc.PiCameraMMALError: no buffers available: Resource temporarily unavailable; try again later_

Attached:
a) the mmal example to play h264 videos :-)
b) the patch for mmalobj.py to make this possible via MMALPythonSource
[mmal_h264_decoder.example.py.txt](https://github.com/waveform80/picamera/files/1636277/mmal_h264_decoder.example.py.txt)
[mmalobj.py.h264.patch.txt](https://github.com/waveform80/picamera/files/1636278/mmalobj.py.h264.patch.txt)

waveform80, 2018-09-28:
> Both video formats are not included in the supported_formats property of the MMALPythonPort class. (I added them manually in my installation to dig deeper in the stuff)

Yup, that's quite deliberate. The MMALPythonSource class is intended as a base class for Python code that produces "something". I've no way of knowing what, so it's up to the developer to override this in their sub-class with something accurate for their source (e.g. H.264). I should make the docs on this a bit more clear (the `mmalobj` docs are still way too low level for serious usage).

> The copy_from() helper raises an error. (may be I added the formats not correctly) source.outputs[0].connect(decoder.inputs[0]) raises an error when source.outputs[0].format is not set explicitly.

I'm afraid the rest of this stems from not quite understanding how to use `mmalobj`; I definitely need to improve the docs in this area with some examples and recipes. Now I've finally got a couple of weekends coming up to work on 1.14 I'll try and squeeze this in.

moritzschaefer, 2018-11-02:
@barney-NG does it play incredibly fast for you as well?

barney-NG, 2018-11-02:
yes.

Dave told me that I didn’t understand the mmal stack  — he’s right :-)

Cheers