waveform80/picamera #472: 6 second exposure takes 50+ seconds to capture?

xoxfaby, 2018-02-27:
following the low light recipe relatively closely, with a shutter_speed of 6000000 which should be 6 seconds, the camere takex about 50 seconds to return an image. 

Interestingly, I actually tried to wait 30 seconds before, and whether or not I wait, the camera always takes the same amount of time, within ~300ms

6by9, 2018-02-27:
Covered many times before in the forums, eg https://www.raspberrypi.org/forums/viewtopic.php?t=164007#p1059785, although 50 seconds sounds a bit high.

The minimum I'm expecting is
- expose and read out one invalid viewfinder frame - 6 seconds
- expose and read out one valid viewfinder frame - 6 seconds.
- stop the sensor and reprogram.
- expose and read out one invalid  capture frame - 6 seconds
- expose and read out one valid capture frame - 6 seconds

So a minimum of 24 seconds to get one capture frame. I've just confirmed that with ```raspistill -ss 6000000 -t 1 -ag 1.0 -dg 1.0 -o foo.jpg``` taking approx 32 seconds to complete, which includes 12 seconds to shut down (it's asked for another preview frame, and has to wait for that to complete).

Seeing as the low light recipe at https://picamera.readthedocs.io/en/latest/recipes1.html#capturing-in-low-light also includes a ```sleep(30)```, is that where your extra delay is coming from?

xoxfaby, 2018-02-27:
Thanks for the extra info.

That's what stumped me, I get 56 seconds whether or not I wait before or not. 

It might be because I'm running the camera in executor since this is in an async application? 

I suppose because I'm waiting in async the camera is doing it's 50 second prep work during the wait?

Is there anything to cut down on that time? Fix values or something? seems like a lot of unnecessary work being done.

6by9, 2018-02-27:
There isn't much you can do to reduce the initial time.
You can use burst mode to reduce the time between captures. It tells the camera not to switch back to viewfinder mode after each capture, therefore saving the time to complete the unwanted viewfinder frame. It'll still take the 12 seconds per capture though.

xoxfaby, 2018-02-27:
12 seconds would be fine but the 56 are kinda insane.

waveform80, 2018-09-28:
Not much I can do here I'm afraid; running the low light example regularly (no async) without the call to `sleep()` results in a similar 32 second capture (basically the same as 6by9's raspistill example). I can't say asyncio (or whatever async layer is popular at the moment) is something I have any intention of supporting given picamera is pretty heavily thread based (and has to be, at least partially, as the thread which feeds picamera data is outside of picamera's direct control). You may be best off using a separate process for the picamera portion and feeding things to your async process via a pipe.

closed 2018-09-28 by waveform80