waveform80/picamera #473: blank preview screen when preview height biger than 698

viki0202, 2018-03-02:
``

viki0202, 2018-03-02:
```
import picamera

with picamera.PiCamera() as camera:

    camera.resolution = (2592, 1944)
        
	camera._preview_fullscreen=0
	camera.preview_window = (0,0,800,699)
	camera.start_preview()
	time.sleep(2)
	camera.stop_preview()
```

6by9, 2018-03-02:
You're exceeding the maximum resolution that the hardware video scalar (the block that puts images on the screen) can cope with. It's specified for 1080P, and only has limited memory for downscaling. If you downscale enough then it line skips as a cheaper downscale which is why it works if the height is <=698.

viki0202, 2018-03-04:
6by9 thanks a lot for quick reply.  Is there any way (workaround) to display the preview in a higher resolution. I've checked the '1080P' and it gives '1080P': (1920, 1080)? Probably my problem is that I still don't understated how the hardware works. I will try to dig deep into specification but any help will be welcome.

6by9, 2018-03-05:
Perhaps the better question is to ask what you are trying to achieve as there may be a better solution.

Video encoding does require preview to run at the same resolution as the encoding, and that will be a problem displaying the images.
Stills encoding has no such requirement and will quite happily switch between resolutions for preview and capture. https://picamera.readthedocs.io/en/latest/api_renderers.html#picamera.PiPreviewRenderer.resolution details the parameter that should allow you to set that up.

As to how the hardware is working, the display engine composes the HDMI output "online", ie it generates the pixels as it needs to send them out to the HDMI display. It therefore has to be able to read the relevant pixels from memory and perform all necessary composition to a very strict timeframe. If you exceed the capabilities then it "underflows" and produces either no output, or a corrupt output. 
Downscaling requires reading two or more lines of image data to produce one line of output, and that therefore requires more memory bandwidth and computation, making it far easier to exceed the capabilities.

waveform80, 2018-09-28:
Looks like 6by9 has answered this pretty comprehensively; do feel free to re-open if you have further questions, but I really can't add any detail beyond what the firmware devs (including 6by9) can (my understanding is a few layers removed from theirs!)

closed 2018-09-28 by waveform80