waveform80/picamera #479: How to remove a stream from memory? (The RAM)

AlexDupont93, 2018-04-06:
I have a motion detection script that uses the Picamera library. I use the PicameraCircularIO stream object, that inherits from the CircularIO class (still a class form the Picamera library), which inherits from the native Python BytesIO class.

The problem is that when recording into the circular stream and then writing the content of the stream on the disk, the memory does not get cleared.

After the stream has been written on the disk, this is what I do:

stream.seek(0)
stream.truncate()

I have also tried calling stream.flush() and/or closing the stream: stream.close(), but it did not help.

I have also tried removing all references to the stream for it to be garbage collected (even though it makes no sense to do that, since truncating the stream to the size 0 should normally do the job...). The stream is in an array, so I did del streams[index]. It did not work, so I also tried calling the garbage collector manually: gc.collect(), with no effect.

It is only the bytes that are being writtin in the stream that are being counted in the memory, not the stream size. So for example, if 2125000 bytes are written into the stream and then I truncate it, there will still be 2125000 bytes in the memory.

Any tip/trick/answer is appreciated.

Cheers!

AlexDupont93, 2018-04-12:
Any idea ? I need to kill my PiCamera script, and restart it to clear the RAM...

AlexDupont93, 2018-05-04:
Anyone have an idea ? I need to kill my PiCamera script, and restart it to clear the RAM...

barney-NG, 2018-05-14:
Why do you want to clear the memory?
The CircularIO guarantees that your stream never exceeds the fixed buffer size. (you defined that at object creation with the size or seconds parameter).
Recording:
Once your stream exceeds the size the old contents of the ring buffer will be overwritten.
Writing:
The last <size> bytes or <number> seconds will be written out. (if  you recorded a full buffer otherwise only the bytes between the start and end pointer will be written out)

AlexDupont93, 2018-05-15:
I am creating a first CircularIO Stream like this : stream = picamera.PiCameraCircularIO(self.camera, seconds = 100, splitter_port=1)
After that there is like 20% of my RAM that is used.
When I am Closing the Camera (self.camera.close()), to reopen it again, I need to create a new CircularIO Stream like the other one : self.stream = picamera.PiCameraCircularIO(self.camera, seconds = 100, splitter_port=1)
After that, my RAM is 40% full.

When I repeat that, my memory got to 100% and leaked... Do you have a solution ?

barney-NG, 2018-05-15:
Hmm -- I need to contact you outside of Github. I just don't know how to do it...

My new question is: Why do you need to close the camera?
Normally you fill the buffer continuously and wait for a specific condition to save.
See:
http://picamera.readthedocs.io/en/release-1.13/recipes1.html#recording-to-a-circular-stream
The next question is:
Do you really need 100 seconds? (Is the event you want to record that long?)

AlexDupont93, 2018-05-15:
Yes I need to close the camera because I am using a multiplexer. So to close the camera correctly I need to camera.close() and reopen it with "camera = picamera.PiCamera(resolution=RESOLUTION, framerate=FRAMERATE)"

For you next question : It is so long because in the ring buffer we have 4 cameras for 25 seconds each.

waveform80, 2018-09-28:
How exactly are you measuring RAM usage? I wouldn't be at all surprised to see the virtual stat staying static even after python objects were freed, but the resident value ought to diminish if the objects are *actually* getting freed. That said, I wonder if there's some circular references involved here (I vaguely recall PiCameraCircularIO keeps a camera reference somewhere). I'll have a look into this for 1.14.

waveform80, 2018-11-03:
> I am creating a first CircularIO Stream like this : stream = picamera.PiCameraCircularIO(self.camera, seconds = 100, splitter_port=1)
> After that there is like 20% of my RAM that is used.
> When I am [...]

I'm afraid I doubt this very much for one simple reason: picamera's `CircularIO` class (which is the parent of `PiCameraCircularIO`) is a bit weird for a ring buffer in that it doesn't pre-allocate any memory. I've added a bit to the docs for 1.14 explaining it in [more detail](https://picamera.readthedocs.io/en/latest/api_streams.html#picamera.CircularIO) but a quick description would be that it's not *really* a ring buffer at all. Instead it operates by keeping references to the buffers written to it in a [deque](https://docs.python.org/3/library/collections.html#collections.deque), dropping references from the "left" when the overall size would exceed the specified limit.

The reason for this weird setup is that it's extremely write-efficient which is what's generally wanted with the camera (we don't copy anything out of the buffer passed to us, we simply keep a reference to that buffer preventing its de-allocation, until it's dropped off the left at which point the GC is free to reclaim it).

Anyway, that basically means that instantiating a `PiCameraCircularIO` takes no (appreciable) memory, no matter how large you request it to be. It'll only start using memory as frames are written to it, and that memory should be freed as soon as chunks drop off the "left" of the deque (unless your code has kept other references to them somewhere).

> That said, I wonder if there's some circular references involved here

I did find some circular references in the structures used in `PiCameraCircularIO` when optimizing a few bits of it during the last week. These wouldn't *prevent* garbage collection, but they'd certainly slow it down (as pure reference counting wouldn't be sufficient to mark them for collection). Given they weren't strictly necessary, I've replaced them with weakrefs for 1.14 (see commit ff9b17a830bf4db4bb716c135b22733d068bad08). After some testing with [tracemalloc](https://docs.python.org/3/library/tracemalloc.html) the stream now de-allocates immediately when falling out of scope (and takes a trivial amount of memory when empty, as before).

closed 2018-11-03 by waveform80