waveform80/picamera #480: how to get actual ISO used when camera.iso=0

jainraje, 2018-04-07:
How can the actual ISO value used by the camera be read when camera.iso is set to 0 (i.e. auto mode)? 

When shutter_speed is set to auto at least the exposure_speed value can be read to get the actual exposure_speed. 

How can the actual ISO used by the camera be read when in auto mode?

TIA,
--Rajeev

simonyangme, 2018-08-02:
I'm also running into this issue and can only get 0 for camera iso

waveform80, 2018-09-28:
There's no attribute provided in MMAL for this (to my knowledge?) so I'm afraid short of querying the EXIF data in a JPEG output there's not much I can do. That said, ISO is *roughly* analogous to gain in the Pi's camera (setting ISO usually fixes the gain to some range of values), so you could query `analog_gain` and `digital_gain` and try and figure out the equivalent ISO from that?

closed 2018-09-28 by waveform80